hzeller/rpi-rgb-led-matrix #1585: Flickering on rpi4 with a new adafruit RGB matrix hat

gguy825, 2023-08-24:
i was having some flickering issues with lines of pixels that have now been fixed with a more powerful power supply and new OS. im running diet pi and no longer get pixel flickering but i do get a random brightness flicker that still continues to persist and im not sure what else to try. Im using 2 64x64 led panels with the additions to the hat like connected 4 and 18 and the soldered panels. my command is 
`sudo ./led-image-viewer -f -s -w3 -C *.png --led-cols=64 --led-rows=64 --led-chain=2 --led-gpio-mapping=adafruit-hat-pwm --led-slowdown-gpio=4 --led-brightness=100`

jaygarcia, 2023-08-29:
@gguy825 , have you performed the following? 

https://github.com/hzeller/rpi-rgb-led-matrix#cpu-use

gguy825, 2023-09-04:
> @gguy825 , have you performed the following?
> 
> https://github.com/hzeller/rpi-rgb-led-matrix#cpu-use

Sorry for late reply. Yea I added that. so my cmdline.txt is currently this. Just making sure this is correct right? 

```
root=PARTUUID=dc9fd960-02 rootfstype=ext4 rootwait net.ifnames=0 logo.nologo console=tty1
isolcpus=3
```

bsparacino, 2024-01-12:
@gguy825 Late reply but my solution has been to overvolt the power supply to 5.1-5.2v