hzeller/rpi-rgb-led-matrix #1587: Multiplexing for non-standard panel: 104*52 res, 1/13 scan rate, 320mm * 160mm LED Panel

andytheengineerguy, 2023-08-28:
Hi,

**TLDR**: I think I've reached the point where I need to make my own multiplexing mapping for this non-standard panel, and was wondering if anyone could lend me a helping hand / point me in the right direction? How do I do this?

**Context**: I would consider myself a medium-level user of this library - I currently have two 128x64 panel pairs running in parallel (basically, two 128*128 duplicate squares) using rpi-rgb-led-matrix, using both C and python bindings, with gifs I have made myself. I wanted a better scan rate so the images would ppear better in photos, so I purchased these 1/13 scan rate panels from Aliexpress (much better than the 1/32 scan rate panels I have now), and am trying to bring them to life. So far.... no luck. All 18 multiplexer mappings, with different row address types, and even different considerations for how they might be 'chained' internally (i.e, 52 cols with a chain of 2, 26 cols with a chain of 4, etc) does not work..... I have tried everything :( So I have accepted I need to make my own multiplexer mapping for this. But immediately, I feel out of my depth.

**Additional Info**:
1. The panel I'm bringing up is this one: https://www.aliexpress.us/item/3256805351469362.html?spm=a2g0o.order_list.order_list_main.5.7b391802I9Ggqy&gatewayAdapt=glo2usa
2. The multiplexing config that's worked the _best_ so far is: 
[image]
3. The rotating square in this case looks like this: 
[image]

andytheengineerguy, 2023-08-30:
Update: Fixed it. Turns out the mapper required for this panel was a form of the ZnMirrorZStripe multiplexing pattern, just with a larger tile_height.

Marking this as closed.

closed 2023-08-30 by andytheengineerguy

bsparacino, 2023-09-08:
@andytheengineerguy Can you share what you did to make this panel work? I believe I have the same panel and am struggling to get it to display properly.

andytheengineerguy, 2023-09-08:
hi @bsparacino, sure, happy to help.  

I found out (with no help from the suppliers or the non-existant datasheet.. 😂!) that the panel _almost_ worked with a ZStripeMultiplexMapper configuration, with even- and odd- block offsets set to 4. But the height was still funky (as shown in the screenshot of my original message).

To fix this, the tile_height needs to be 13 (not 4, which is the original value), as shown:
[image]

So...
1. Change height to 13
2. Register a new multiplexer mapper [image]
3. Re-make the codebase
4. Run the rotating square demo using --led-multiplexing=19

Once you've confirmed this solution actually works, I'd recommend modifying the ZStripeMultiplexMapper class in order to preserve the existing mappings that depend on it. I achieved this by adding the width and height as variables that you pass _into_ the constructor. So then we can pass our new height of 13 while still utilizing the existing ZStripeMultiplexMapper class.
[image]

When you do this, don't forget to update all the other registrations with their default tile sizes (8 and 4)
[image]

Come to think of it I will probably do a PR for this change.

Good luck, lmk how you go.

bsparacino, 2023-09-08:
Not working for me, I will play around with the height though and maybe something will work
What are the other parameters you are using?
`./demo -D 0 --led-gpio-mapping=adafruit-hat --led-cols=104 --led-rows=52 --led-row-addr-type=0 --led-multiplexing=19`

[image]

bsparacino, 2023-09-08:
What parameters are you using to run the demo? When I run it with multiplexing 5 it is similar to your initial image. But with your code added I am not ending up with the corrected image.

`./examples-api-use/demo -D 0 --led-gpio-mapping=adafruit-hat --led-cols=104 --led-rows=52  --led-row-addr-type=0 --led-multiplexing=19`
[image]

`./examples-api-use/demo -D 0 --led-gpio-mapping=adafruit-hat --led-cols=104 --led-rows=52  --led-row-addr-type=0 --led-multiplexing=5`
[image]

andytheengineerguy, 2023-09-08:
Hmm. Strange. Your panel looks the same in terms of specs (resolution, scan rate, size) but the physical construction is definitely different than mine. Different locations of the IDC connector (maybe different LED driver chips?) etc.

We are using different electronics. I'm using an RPi4 with an active-3 hat from electrodragon.

I just ran the demo with the following and it worked `./demo -D 0 --led-gpio-mapping=regular --led-cols=104 --led-rows=52 --led-slowdown-gpio=4 --led-row-addr-type=0 --led-multiplexing=19`
[image]

Just re-ran the 5 pattern, we are definitely getting _similar_ results for 5 `./demo -D 0 --led-gpio-mapping=regular --led-cols=104 --led-rows=52 --led-row-addr-type=0 --led-multiplexing=5 --led-slowdown-gpio=4`
[image]

andytheengineerguy, 2023-09-08:
I have not used the adafruit hat before. The only other thing I can think of right now is the connector type - is it HUB75 or HUB75E? If it's E, you might need to enable the E line. I needed to do that with the 1st panels I brought up using this library.

But for these 104*52s specifically, they are regular HUB75s, so I didn't need to worry about it.

bsparacino, 2023-09-08:
About 20min ago I actually soldered the E line on the hat which resulted in the photos I posted, before I did that it was a total mess.

bsparacino, 2023-09-08:
Well I must have different modules, this function solved it for me
https://github.com/hzeller/rpi-rgb-led-matrix/issues/1499#issuecomment-1385945046

andytheengineerguy, 2023-09-08:
Out of curiosity I just tried the mapper from [#1499 (comment)](https://github.com/hzeller/rpi-rgb-led-matrix/issues/1499#issuecomment-1385945046) and got a similar vertically-jumbled pattern to what you started with. So, yep, different panels indeed. At least its consistent:
[image]

Well, glad you got yours working, and good to know that if I ever need an alternative for these 1/13, 104*52 panels, I know which mapping to use 💪