hzeller/rpi-rgb-led-matrix #1592: Outdoor Matrix, Segmented output issue

ValentinLaulund, 2023-09-24:
Hi all

The older outdoor matrix module we use, works great but has been discontinued, so I am trying to find a new P3 outdoor module.

The new I have found so far has these 2x4 segments split issue, it also look like the input is downscaled or duplicated, see image below (same hardware setup)     

[image]

I have tried, all combinations of the, --led-multiplexing. --led-row-addr-type --led-panel-typ
Also swapped around the .a .b .c .d .e, it just changes the colors ... Surprise! ;) 

My command string have been something like:
sudo ./demo -D0 --led-rows=64 --led-cols=64 --led-multiplexing=1 --led-gpio-mapping="adafruit-hat-pwm"

I hope one of you guys have inputs for what I can try out, or tips on where to acquire an Outdoor (Brightness >= 5000 cd/m2) P3 64x64 RGB LED Matrix (192x192) that works with an Raspberry PI and the rpi-rgb-led-matrix library.

anandrajgupta, 2023-09-25:
did you try  --led-multiplexing=0

ValentinLaulund, 2023-09-25:
> did you try --led-multiplexing=0

Yeah I tried all of the 18 options ;(

closed 2025-11-27 by marcmerlin