hzeller/rpi-rgb-led-matrix #1593: P10 Outdoor Panel 1/4 Scan Problem

Xaver88, 2023-10-01:
Hi there
got a P10 outdoor 1/4 scan 32x16 display and can't get ist work
it seems that every 4 rows are not in the correct order

https://github.com/hzeller/rpi-rgb-led-matrix/assets/105126414/1a8a0b29-2384-444a-be8f-65a4b48021af

anandrajgupta, 2023-10-01:
multiplexing?

Xaver88, 2023-10-01:
tried all from 0 to 18, but no success
this example is 3

Xaver88, 2023-10-01:
for me it seems that row 0-3 and 4-7 are swapped
also 8-11 and 12-15
same an the second panel
but how to solve this?

anandrajgupta, 2023-10-01:
your problem will be solved after a minor tweaking in led-multiplexer.cc file

Xaver88, 2023-10-01:
I'm not firm with cc
can you give me a hint or example what to do
when I change the led-multiplexer.cc, what's next to get the demo (and python) workung)
do I have to "make" the libs again?

Xaver88, 2023-10-01:
there is no file led-multiplexer.cc
which do you mean?

anandrajgupta, 2023-10-02:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/multiplex-mappers.cc

yes you have compile the libraries again

Xaver88, 2023-10-02:
tried to change and complie it. Output on the display changes, but (at the moment) I don't understand how the multiplex mapper and the pixelmapper works

anandrajgupta, 2023-10-02:
trial and error

map each x,y coordinates

davemaster, 2023-10-02:
> Hi there got a P10 outdoor 1/4 scan 32x16 display and can't get ist work it seems that every 4 rows are not in the correct order
> VID-20231001-WA0010.mp4

Greetings,

What command are You trying? ex. sudo demo --led-cols=32 --led-rows=16 --led-multiplexing=8 -D 0 (the 8 works for my 32x16 1/4 scan outdoor P10)

Xaver88, 2023-10-02:
> Greetings,
>
> What command are You trying?
>
> **ex. sudo demo --led-cols=32 --led-rows=16 --led-multiplexing=8 -D 0** (the 8 works for my 32x16 1/4 scan outdoor P10)

I tried all combinations and the mutliplexing=3 or 13 brings the shown video
also asked the manufacrurer for a pixelmapping but get no answer (until now)

davemaster, 2023-10-02:
> I tried all combinations and the mutliplexing=3 or 13 brings the shown video also asked the manufacrurer for a pixelmapping but get no answer (until now)

Can You please take a picture of the back of rgb panel?

anandrajgupta, 2023-10-03:
led-multiplexing=4 worked for my P10 4 Scan 16×32.
May try tweaking other parameters keeping this multiplexer=4

Xaver88, 2023-10-05:
get it work with a little twaek in the mapping
but how to compile (MAKE ?) that also the python examples will work now?

anandrajgupta, 2023-10-05:
i am not about the python one, i used C

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828