hzeller/rpi-rgb-led-matrix #1595: How to pixel mapping divide chain in two

Ry-Kode, 2023-10-04:
Hello,

I want to drive a 4 x 4 matrix display using two chains of 8 panels, but I can't find the correct pixel-mapping for straight connected cabling.

This is the cabling:

[image]

I do not want to use U-mapper because this gives two dark horizontal bands in the image caused by the panels that are upside-down (rows 2 and 4).

Thanks!

anandrajgupta, 2023-10-05:
https://github.com/hzeller/rpi-rgb-led-matrix/issues/1471

xuniuer, 2023-10-07:
You should use `U-mapper` mode with 8 chains and two parallels.

anandrajgupta, 2023-10-07:
He doesn't want the panels to be upside-down

Ry-Kode, 2023-10-07:
Thank for the link @anandrajgupta, I'll see where this gets me.