hzeller/rpi-rgb-led-matrix #1598: Formatting leads to library conflicts

MontgomeryNoth, 2023-10-17:
Dear all,
while trying to build a stock-price ticker I ran into an 'OperationalError('unable to open database file') as soon as I use the graphics function (named Panel1 in the example code). My experience with the package is very limited and it seems like the yfinance library does not work without errors while the current graphics functions are used, I would be happy for any hints/help. Thanks!

### Replication code

```
import yfinance as yf
import time
import pandas as pd
from datetime import date, timedelta
from pandas.tseries.offsets import BDay

from rgbmatrix import RGBMatrix, RGBMatrixOptions, graphics

def FormatIntraday(intraday, Results, Tickers):
    # Intraday formatting function
    Results['Price'] = intraday[::-1].bfill().iloc[0]
    Results['DOD'] = (Results['Price']/monthly.loc[date.today() - BDay(1),:] -1)*100

def Panel1(matrix, canvas, Input):
    for i in range(0, len(Input)):
        if Input.iloc[i,2]<0:
            dod_color = graphics.Color(0, 0, 255)  # Red
        else:
            dod_color = graphics.Color(0, 255, 0)  # Green
        if Input.iloc[i,3]<0:
            mtd_color = graphics.Color(0, 0, 255)  # Red
        else:
            mtd_color = graphics.Color(0, 255, 0)  # Green 

        fontlarge = graphics.Font()
        fontsmall = graphics.Font()

        fontlarge.LoadFont("../../../fonts/7x13.bdf")
        fontsmall.LoadFont("../../../fonts/5x7.bdf")

        
        graphics.DrawText(canvas, fontlarge, 0, 9+i*16, graphics.Color(255, 255, 255), str(Input.iloc[i,0]))
        graphics.DrawText(canvas, fontsmall, 64-5*len(str(round(Input.iloc[i,1]))), 9+i*16, mtd_color, str(round(Input.iloc[i,1])))
        graphics.DrawText(canvas, fontsmall, 0, 15+i*16, dod_color, str(round(Input.iloc[i,2],2)))
        graphics.DrawText(canvas, fontsmall, 64-5*len(str(round(Input.iloc[i,3],2))), 15+i*16, mtd_color, str(round(Input.iloc[i,3],2))) 

Tickers = ['VWCE.DE', 'JREG.DE', 'XDEB.DE', 'IQQW.DE','AGN.AS', 'DTG.DE', 'CALN.SW', 'XUHY.DE']

monthly = yf.download(Tickers, start=date.today() - timedelta(31) + pd.offsets.BMonthEnd())['Close']

Matrixoptions = RGBMatrixOptions()
Matrixoptions.rows = 64
Matrixoptions.cols = 64
Matrixoptions.chain_length = 3
Matrixoptions.parallel = 1
Matrixoptions.hardware_mapping = 'regular' 

matrix = RGBMatrix(options = Matrixoptions)

try: 
    while True:       
        print("Press CTRL-C to stop.")
        offscreen_canvas = matrix.CreateFrameCanvas()

        Results     = yf.download(Tickers, period='1d', interval='1m')['Close']
        
        offscreen_canvas.Clear()
        # Graphics formatting
        Panel1(matrix, offscreen_canvas, Results)
        
        offscreen_canvas      = matrix.SwapOnVSync(offscreen_canvas)
        
        time.sleep(10)
        
        
except KeyboardInterrupt:
    # Handle a keyboard interrupt (Ctrl+C) to exit the loop gracefully
    print("Keyboard interrupt received. Exiting the loop.")
```

combs, 2024-03-08:
this is the github repo for the LED matrix library--your problem is inside `yfinance`

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828