hzeller/rpi-rgb-led-matrix #1600: Controlling RGB LED display with Orange Pi 3B GPIO and Rockchip rk3566

SturdyCat, 2023-11-04:
[Controlling RGB LED display with Orange Pi 3B GPIO and Rockchip rk3566](https://github.com/SturdyCat/opi-rockchip-3b-rgb-led-matrix)

652436962, 2023-11-09:
> [Controlling RGB LED display with Orange Pi 3B GPIO and Rockchip rk3566](https://github.com/SturdyCat/opi-rockchip-3b-rgb-led-matrix)

this is really ?   this is readme about raspberry pi

SturdyCat, 2023-11-11:
> > [Controlling RGB LED display with Orange Pi 3B GPIO and Rockchip rk3566](https://github.com/SturdyCat/opi-rockchip-3b-rgb-led-matrix)
> 
> this is really ? this is readme about raspberry pi

On my orange pi rk3566 works fine

652436962, 2023-11-13:
> > > [Controlling RGB LED display with Orange Pi 3B GPIO and Rockchip rk3566](https://github.com/SturdyCat/opi-rockchip-3b-rgb-led-matrix)
> > 
> > 
> > this is really ? this is readme about raspberry pi
> 
> On my orange pi rk3566 works fine

which  os?   Do you a guide markdown?

652436962, 2023-11-16:
> > > [Controlling RGB LED display with Orange Pi 3B GPIO and Rockchip rk3566](https://github.com/SturdyCat/opi-rockchip-3b-rgb-led-matrix)
> > 
> > 
> > this is really ? this is readme about raspberry pi
> 
> On my orange pi rk3566 works fine

Do you have wechat or qq

SturdyCat, 2023-11-18:
> > > > [Controlling RGB LED display with Orange Pi 3B GPIO and Rockchip rk3566](https://github.com/SturdyCat/opi-rockchip-3b-rgb-led-matrix)
> > > 
> > > 
> > > this is really ? this is readme about raspberry pi
> > 
> > 
> > On my orange pi rk3566 works fine
> 
> Do you have wechat or qq

<!DOCTYPE html><h2 tabindex="-1" dir="auto" style="box-sizing: border-box; margin-top: 24px; margin-bottom: 16px; font-size: 1.5em; font-weight: var(--base-text-weight-semibold, 600); line-height: 1.25; padding-bottom: 0.3em; border-bottom: 1px solid var(--borderColor-muted, var(--color-border-muted)); caret-color: rgb(230, 237, 243); color: rgb(230, 237, 243); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-style: normal; font-variant-caps: normal; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-tap-highlight-color: rgba(26, 26, 26, 0.3); -webkit-text-size-adjust: 100%; -webkit-text-stroke-width: 0px; text-decoration: none;">Wiring it up</h2><p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; caret-color: rgb(230, 237, 243); color: rgb(230, 237, 243); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 16px; font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-tap-highlight-color: rgba(26, 26, 26, 0.3); -webkit-text-size-adjust: 100%; -webkit-text-stroke-width: 0px; text-decoration: none;">[image]</p>
HUB75E | PIN | GPIO4 | HUB75E | PIN | GPIO4
-- | -- | -- | -- | -- | --
16 | R1 | 140 | 15 | G1 | 141
14 | B1 | 147 | 13 | GND |  
12 | R2 | 128 | 11 | G2 | 130
10 | B2 | 138 | 9 | E | 136
8 | A | 139 | 7 | B | 131
6 | C | 129 | 5 | D | 132
4 | CLK | 134 | 3 | LAT | 135
2 | OE | 144 | 1 | GND |  

<p dir="auto" style="box-sizing: border-box; margin-top: 0px; margin-bottom: 16px; caret-color: rgb(230, 237, 243); color: rgb(230, 237, 243); font-family: -apple-system, BlinkMacSystemFont, &quot;Segoe UI&quot;, &quot;Noto Sans&quot;, Helvetica, Arial, sans-serif, &quot;Apple Color Emoji&quot;, &quot;Segoe UI Emoji&quot;; font-size: 16px; font-style: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: auto; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: auto; word-spacing: 0px; -webkit-tap-highlight-color: rgba(26, 26, 26, 0.3); -webkit-text-size-adjust: 100%; -webkit-text-stroke-width: 0px; text-decoration: none;">Orange Pi needs to shut down the kernel<span class="Apple-converted-space"> </span><code style="box-sizing: border-box; font-family: ui-monospace, SFMono-Regular, &quot;SF Mono&quot;, Menlo, Consolas, &quot;Liberation Mono&quot;, monospace; font-size: 13.6px; padding: 0.2em 0.4em; margin: 0px; white-space: break-spaces; background-color: var(--bgColor-neutral-muted, var(--color-neutral-muted)); border-radius: 6px;">CONFIG_STRICT_DEVMEM</code><span class="Apple-converted-space"> </span>for<span class="Apple-converted-space"> </span><code style="box-sizing: border-box; font-family: ui-monospace, SFMono-Regular, &quot;SF Mono&quot;, Menlo, Consolas, &quot;Liberation Mono&quot;, monospace; font-size: 13.6px; padding: 0.2em 0.4em; margin: 0px; white-space: break-spaces; background-color: var(--bgColor-neutral-muted, var(--color-neutral-muted)); border-radius: 6px;">mmap</code><span class="Apple-converted-space"> </span>to work.</p>

phongans, 2023-12-21:
I can't play video on Orange Pi 3B with [Controlling RGB LED display with Orange Pi 3B GPIO and Rockchip rk3566](https://github.com/SturdyCat/opi-rockchip-3b-rgb-led-matrix). Please help explan how to runing this?

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin