hzeller/rpi-rgb-led-matrix #1603: Rasberry Pi 5 support?

huntercaron, 2023-11-16:
I recently got a Pi5 and was hoping to set up these panels I have, but don't seem to be getting any LEDs to light up whatsoever. 

I'm trying with a single 64x64 1/32 panel with a IC2037BP chipset, connected directly via wires. ([This Panel](https://www.aliexpress.us/item/4000365937036.html)).

- I've hooked up the E pin
- I've tried `led-slowdown-gipio=4`

From what I’ve read in the issues here, it should be working pretty smoothly. So I am starting to think its something with the Pi5? Happy to give more info here, just not sure what would be helpful

hzeller, 2023-11-17:
I have not have adapted the library yet to work with Pi5. Since there were _always_ additions needed with every new Pi, as each GPIO is different, it is no surprise that the new one is not working yet.

Good news: I got a Pi5 so I can work on that.
But bad news: I don't know yet when I'll have time to adapt the library, might be probably around the end of the year. So if you have, use an older Pi for now.

huntercaron, 2023-11-17:
Ah makes sense! Thank you for the quick response.

I think I have an older Pi somewhere I will dig up in the meantime. Please let me know if you need any help testing.

huntercaron, 2023-11-20:
_Update_: All is working well with Pi 4 for now

babsher, 2023-11-28:
@hzeller I am also interested in Pi5 support. Is this in the works still?

Primek17, 2023-12-09:
Hi @hzeller. Is there a chance to launch the library on PI5 this month? :)

imCharlieB, 2024-01-07:
Any updates on the pi5 support?

bsparacino, 2024-01-25:
I am trying to adapt the code to get the Pi 5 to work but am struggling, maybe someone smarter can figure it out.
I simply followed the process for how other models of the Pi are detected.
The thing I seem to be struggling with is the value for BCM2712_PERI_BASE.

It's possible the Pi 5 drastically changed how its offsets worked, but I can't seem to find a datasheet on the BCM2712 and even if I did, I would still be too confused to know what to do.

My code is here:
https://github.com/hzeller/rpi-rgb-led-matrix/compare/master...bsparacino:rpi-rgb-led-matrix:master

davemaster, 2024-01-25:
I will try tomorrow just as I do with a RPi 4

imCharlieB, 2024-01-25:
i found this little bit of info browsing cant rem what site i saved it in a note so i could look at it later...

 The Pi 5 has a huge peripheral area starting at 0x40000000. This is the physical address which is translated to 0x1f00000000 in the 40-bit address space. The /dev/mem file can be used in the usual way but you have to be running with root privileges. The /dev/gpiomem0 file only maps the GPIO area starting at 0x400d0000, or 0x1f000d0000, but it doesn’t need root privileges. In the examples that follow, the mem file is used with root privileges because it works with peripherals other than the GPIO lines.

To access the GPIO area of memory you can map the entire peripherals area into user space using:

 int memfd = open("/dev/mem", O_RDWR | O_SYNC);
    uint32_t *map = (uint32_t *)mmap(
        NULL,
         64 * 1024* 1024,
        (PROT_READ | PROT_WRITE),
        MAP_SHARED,
        memfd,
        0x1f00000000
 );
    if (map == MAP_FAILED)
    {
        printf("mmap failed: %s\n", strerror(errno));
        return (-1);
    };
    close(memfd);

 

This maps the entire 64MByte area – if you only want to use the GPIO lines then you can make this smaller, but there is no problem accommodating it in the 4,096MByte, 32-bit address space.

Once you have the address of the peripheral area you can access the registers that control them using offsets from the hardware-specified base address. For example, the GPIO control registers have an address of 0x400d0000, which gives an offset of 0xd0000. As the pointers are to uint32_t we have to remember to divide by 4 so that the correct offset is added. That is, the GPIO registers are located at:

    uint32_t *PERIBase = map;
    uint32_t *GPIOBase = PERIBase + 0xd0000 / 4;

IsaacMAllen, 2024-01-29:
> i found this little bit of info browsing cant rem what site i saved it in a note so i could look at it later...
>
> The Pi 5 has a huge peripheral area starting at 0x40000000. This is the physical [...]

Here's the link:
https://www.i-programmer.info/programming/148-hardware/16887-raspberry-pi-iot-in-c-pi-5-memory-mapped-gpio.html

DJEF97, 2024-02-03:
> Hi @hzeller. Is there a chance to launch the library on PI5 this month? :)

bsparacino, 2024-02-04:
I'm still not having luck trying this.
I am also not 100% sure what to do with this info
> uint32_t *PERIBase = map;
> uint32_t *GPIOBase = PERIBase + 0xd0000 / 4;

JimM58, 2024-02-20:
Ok, I just got a pi5 running bookworm, and trying to run some apps on it.  In this case, the adafruit-circuitpython-rgb-display stuff 

And I get this

pi@pi5:~/ciq-common $ python3 tfteco.py
Traceback (most recent call last):
  File "/home/pi/ciq-common/tfteco.py", line 209, in <module>
    cs_pin = digitalio.DigitalInOut(board.CE0)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/digitalio.py", line 185, in __init__
    self.direction = Direction.INPUT
    ^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/digitalio.py", line 215, in direction
    self._pin.init(mode=Pin.IN)
  File "/usr/local/lib/python3.11/dist-packages/adafruit_blinka/microcontroller/generic_linux/libgpiod/libgpiod_pin_1_x.py", line 92, in init
    self._line.request(
OSError: [Errno 16] Device or resource busy

It runs fine on all earlier pi devices.

Here's what I have on the pi 5 as far as adadfruit packages:

Package                            Version
---------------------------------- ----------
Adafruit-Blinka                    8.32.0
adafruit-circuitpython-bme280      2.6.24
adafruit-circuitpython-busdevice   5.2.6
adafruit-circuitpython-requests    2.0.5
adafruit-circuitpython-rgb-display 3.11.3
adafruit-circuitpython-typing      1.10.0
Adafruit-PlatformDetect            3.60.0
Adafruit-PureIO                    1.1.11

hannescam, 2024-02-22:
> [IsaacMAllen](https://github.com/IsaacMAllen) I'm still not having luck trying this. I am also not 100% sure what to do with this info
> 
> > uint32_t *PERIBase = map;
> > uint32_t *GPIOBase = PERIBase + 0xd0000 / 4;

Do you have any progress with the Raspberry Pi 5 support?
Maybe this will help: [opi-rockchip-3b-rgb-led-matrix](https://github.com/SturdyCat/opi-rockchip-3b-rgb-led-matrix)

0xe1f, 2024-02-28:
Same as all others - unable to run on Pi 5:
```
mmap error: : Invalid argument
                                                                                                                               MMapping from base 0x3f000000, offset 0x3000
mmap error: : Invalid argument
MMapping from base 0x3f000000, offset 0x101000
Need root. You are configured to use the hardware pulse generator for smooth color rendering, however the necessary hardware registers can't be accessed because you probably don't run with root permissions or privileges have been dropped.
So you either have to run as root (e.g. using sudo) or supply the --led-no-hardware-pulse command-line flag.
	Exiting; run as root or with --led-no-hardware-pulse
```

I am running as root.

MikeDaBird, 2024-04-03:
Bumping this. Currently working on a project utilizing chained Adafruit 64x32 matrices, but so far cannot get a display on one using Pi 5. It could be my current power situation, but I feel like the compatibility thing could also be a culprit. It's weird (and kind of misleading) that Adafruit's description and docs on their Bonnet HAT say they support _"any Raspberry Pi with a 40-pin GPIO header"_ and explicitly list Pi 5 in the description, yet they list the use of this library which does not support Pi 5 :/ (supporting this: their installer wrote the _dtparam=audio=off_ line for the Quality option in the old config.txt location prior to Bookworm OS, which Pi 5 requires)

I did get a Pi 3 as a backup and haven't tested the bonnet on it yet, but it can barely run what I need it to run, so preferably would use the Pi 5 once possible unless I can figure out how to optimize my software even more.

Pretty new to Pi, so I'm unable to really troubleshoot except through a lot of Google searching. Hopefully you can manage to find time to figure out support for it!

AbrahamChalita, 2024-04-08:
Hello! Has anyone had any luck with the Pi 5 support?

Vizzyy, 2024-04-16:
I also ran into the above output @0xe1f mentioned above with a new rpi5. 

I dug out an older rpi3 I have, slapped on the hat, swapped the SD cards and my 64x32 matrix worked instantly with the demo script. 

It's unfortunate that as @MikeDaBird noted the Adafruit website explicitly lists support with rpi5, but we're not quite there yet.

ladyada, 2024-04-23:
oops sorry about that, it was a copy-paste typo (since almost all of our HATs do work with the Pi 5) - if you cant use it email support@adafruit for a refund :)

MikeDaBird, 2024-04-23:
Heya, thanks for acknowledging and fixing the description. I did get a Pi 3 for backup as mentioned above so I can still work on my project. Did get both my Pi 3 and 5 from the Adafruit website in the same order, and will keep both in the mindset that Pi 5 support will eventually be complete (with that big graphics improvement it would be really nice to have support for it). Besides, I can experiment with other things using my spare Pi ^^

ignaciogaticarojas, 2024-04-29:
upgrading this program to work also with raspberry pi 5 would be an amazing option for it! hopefully @hzeller would have some time to work on it . Will keep an eye on this thread

ElBertrando, 2024-07-09:
@hzeller I just got a bunch of Pi5's and have my program ready to load ! Do we have an ETA ?

sawsaw382, 2024-07-12:
Can I add to the above, very keen to get Pi5 compatibility enabled.

hzeller, 2024-07-12:
I've reserved a bit of time this weekend to work on it.

bsparacino, 2024-07-12:
@hzeller 
This is as far as I was able to get on it, it may save you a little work
https://github.com/hzeller/rpi-rgb-led-matrix/compare/master...bsparacino:rpi-rgb-led-matrix:master

Speedy-VI, 2024-07-13:
I have not had any luck finding P2 128x64 panels that use LED driver chips that will work with the Matrix Portal S3. I'm not saying it's impossible, but I have only had success with lower-density panels that use ICN2037, FM6124, or RUL2038S driver chips.  It seems that to control panels with more complex driver chips, I need a Raspberry Pi. I was about to take the plunge and order a Pi 4 B but the Pi 5 is only $5 more.  If there is one thing I have learned from buying LED panels on Ali Express, it's patience! Fingers crossed that support for the Pi 5 is coming soon.

hzeller, 2024-07-15:
### Update on the Pi 5 exploration

I did get the memory mapping and parallel output on the GPIO pins to work in my local experiments, so the first hurdle is taken. However, it is slower than other recent Raspberry Pi models.

The GPIO handling is _very_ different on the Pi5 compared to the earlier models. Previously, it was handled directly from the SoC, now the Pi5 handles that with a [separate chip](https://www.raspberrypi.com/documentation/microcontrollers/rp1.html) that is connected via the PCIe system bus. Due to that, it can't reach the same performance.

There are still are few more hardware things to figure out (in particular hardware timers and routing them to GPIO) before rpi-rgb-led-matrix can be updated. Unfortunately, the documentation by Raspberry is in parts _very_ incomplete, so I have to fish out relevant information from sources on the raspberrypi github as well as the Linux kernel. This might take a while.

What I can say: output will probably not reach higher frame-rates (and likely be lower) than with a Pi 4 or Pi 3, so if you want to use this library, I don't currently recommend hoping for the Pi5 to be faster once support has been added. Get a Pi4 for now.

0xe1f, 2024-07-15:
Thanks for taking the time to look into it. I ultimately decided to use multiple Pi4's, and am able to drive 320x128 just on a single Pi4.

Speedy-VI, 2024-07-15:
@hzeller
Thanks for the update. I was surprised to learn that the Pi 5 will likely be slower than a Pi 4 due to offloading GPIO to the RP1 southbridge chip. I think I will order a Pi 4 Model B.

Newbie Question - I did not see anything in the README about RAM requirements. The Pi 4B is available with up to 8-Gig of RAM. Will more RAM make any difference when using a Pi 4B and this library to drive LED panels?

primaltechguides, 2024-09-09:
I'm sure I got it working with my pi5 a few months ago.

I remember having to set the GPIO slowdown to like 20 or 30, I had to mod the source as there was a max limit. I was using PWM as well btw if that helps.

hzeller, 2024-09-09:
send a PR

primaltechguides, 2024-09-13:
Would this be relevant to the implementation?
From the WiringPi repositiory

https://github.com/WiringPi/WiringPi

Pi5 full PWM support
https://github.com/WiringPi/WiringPi/commit/f2b23cfb0dbc8a988b8929aaacf8245047dbb047

Krasnomakov, 2024-10-01:
> ### Update on the Pi 5 exploration
> I did get the memory mapping and parallel output on the GPIO pins to work in my local experiments, so the first hurdle is taken. However, it is slower than other [...]

Hello, are there any updates? Info on what is possible now?

davemaster, 2024-10-01:
> Hello, are there any updates? Info on what is possible now?

May be that, it's a GOOD THING, I imagine that GPIO "section" now can refresh data to panel(s) meanwhile the main "section" of the SBC can manage other matters, or am I wrong? 

Thanks for your research

bitslip6, 2024-10-18:
PI5 support for HUB75 is available here with GPU rendering (not for GPIO, just frame buffer rendering suing fragment shaders)
The implementation is a bit different than this library but it should have enough info to port to rpi-rgb-led-matrix if someone is inclined.  It ONLY supports Pi5 as I still need to track down the GPIO mmap addresses for pi4 and pi3.

https://github.com/bitslip6/rpi-gpu-hub75-matrix

ydixken, 2024-10-25:
Also struggled with this a lot, before ich found this bug. Support would be highly appreciated!

bitslip6, 2024-10-25:
@ydixken 
This library is similar with pi5 support. drop an issue here and I'll take a look: https://github.com/bitslip6/rpi-gpu-hub75-matrix/issues

ydixken, 2024-10-25:
@bitslip6 Thanks, just tried it out, can't get it to build on a fresh OS.

marcmerlin, 2024-11-25:
@Robyn-bot  Henner has not had the time/resources to work on this. 
Between you and me, a Pi4 is cheaper and already works faster than you can output 3 channels, so you need to slow it down.
There is no reason to use a Pi5 or great pressing need to make this lib on an even faster and more expensive board just that it can be slowed down even more not to overwhelm the panels with too much I/O speed.

marcobrianza, 2024-12-27:
hello, I have been a long time user of this software, and today after this blog post https://blog.adafruit.com/2024/12/26/driving-hub75-rgb-matrices-on-raspi-5-with-pio/
I came here to check pi5 support. maybe Adafruit work can be beneficial :-) ciao!

marcmerlin, 2024-12-27:
@marcobrianza The adafruit HUB75 support has always been super basic. It's cool that someone hacked something on Pi5, but that work doesn't translate in any way to this lib. This lib would have to be seriously adapted to the new hardware on Pi5 and last I talked to @hzeller  just a month ago, he had no plans to work on it at this time, since he has no personal need.
I will also add again that a Pi4 is already faster than needed for 3 channels at speeds faster than the panels can accept, so using a Pi5 is just paying more money for faster hardware that has to be slowed down even more, and with software support that does not exist as of today.
So again, the question is why?
If you find it fun to port it, because why not, please do so and contribute back.
If you are expecting existing users to have enough of a need for this to spend their time to do it, I'd say the odds are low since as I said again Pi4 is cheaper, already too fast, and works today.

marcobrianza, 2024-12-27:
>why?
for my experience on pi4 with all the recommended optimisation the system is often on the edge to glitch if using the board for simultaneous tasks. (for this reason I have mostly switched to colorlight cards driven by fpp)
I expect that the Pi5 using the pio external chip could optimise led driving decoupled from the cpus
From what I understand current achievements of Adafuits are due to a recent raspberry pi releasing some examples previously unavailable.

marcmerlin, 2024-12-28:
@marcobrianza fair enough.
Note that glitches are due to linux not being real time (although apparently real time was just merged into upstream kernels very recently) and therefore any GPIO bit banging can be glitchy indeed, no matter how fast the CPU is (i.e. it will be mostly the same on RPI5 as long as the code is using the CPU to push bits).
Trust me, I know, I have very fast many core CPUs computers doing some bit banging for some old protocols, and they routinely have issues when the system is too busy/stuck with IO and the other threads are blocked.
The only way to make this totally glitch free would indeed be to write some full DMA driver using some dedicated IO pins, which may exist on Pi5, but that would indeed be a new driver backend to write from scratch.
As you already pointed out, if you need perfect forever glitch free output, FPGA boards are indeed the way to go

bitslip6, 2025-01-19:
> @marcobrianza The adafruit HUB75 support has always been super basic. It's cool that someone hacked something on Pi5, but that work doesn't translate in any way to this lib. This lib would have to [...]

This is incorrect. The reason that the delays need to be added is not for the hub75 panels but because the horribly rounded edges on the rpi 4 "square" wave form.  This is a hardware limitation on thr pi4 not on thr panels themselves whixh you can see are driven beautifully by better hardware.  There is pi5 support on another library in this thread with 10x performance (20Mhz, vs 2Mhz on rp4).  Piolib could theorirtically push this to 70Mhz.

ValentinR06, 2025-01-30:
Is there any Update on the Raspberry Pi5 support?

ladyada, 2025-01-30:
we have a separate project to add HUB75 via PIO to Pi 5 https://github.com/adafruit/Adafruit_Blinka_Raspberry_Pi5_Piomatter
its not documented yet but it does work

marcmerlin, 2025-01-30:
Thanks @ladyada , much appreciated.
@bitslip6 thanks for correcting my understanding on maximum speed limits. It's good to know that we could output at a higher speed if the square waveform were more square.
@ladyada , is the rPi5 port using IO hardware that is more precise and more square, like an FPGA output?

ladyada, 2025-01-30:
yes PIO is a high speed peripheral, it does something like 150 FPS and

marcmerlin, 2025-01-30:
This is great news, thank you @ladyada .   As for IO pins available, how many channels can be controlled by those new PIO pins, and how many are currently supported by the driver you mentioned? (looks like one, but just checking).
Reason I ask is that if it's just one channel, I'm going to guess that for a big display, it may still be faster with an rPi4 and 3 channels than an rPi5 and one channel (which would divide bandwidth by 3 and add additional issues with clock mis-sync on long bus wiring)

ladyada, 2025-01-30:
pretty sure you could just use another PIO state machine for another panel.
we havent tried or inplemented it, if someone wants to look into it.

bluelasers, 2025-02-13:
Concern is for stability and correctness of multiplexing with remote IO.  [This](https://github.com/hzeller/flaschen-taschen) should provide a starting point for managing stability. The RP1 does offer better clock management. Using packet mode via RP1 should improve performance. The forces involved in stability are not always obvious to the user or programmer.

I would recommend the RP1 using PIO for most projects once supported. Piomatter may require an internal refactor.

bitslip6, 2025-03-05:
Thanks @ladyada for that great repo. I'd love to add rp1 support to https://github.com/bitslip6/rpi-gpu-hub75-matrix. I took a look at the source. it's pretty lightly commented. Do you have a link to a programming reference for the RP1? I have read https://datasheets.raspberrypi.com/rp1/rp1-peripherals.pdf but I don't see any details on interfacing from linux user or kernel land to the RP1. Can you send link to any additional programming references?

ladyada, 2025-03-05:
https://www.raspberrypi.com/news/piolib-a-userspace-library-for-pio-control/

sub-zerohm, 2025-03-06:
Well time for me to learn some python.  I was dreading it but thank you @ladyada , I'm super psyched to be able to do this because the pi 5 has its own fan pins, Don't need to have a fan on your Matrix controller hat

bitslip6, 2025-03-11:
What is the bottleneck for this application? It seems like we can achieve higher performance with multi threading and an optimized loop toggling BCM data via GPIO? Obviously hardware sync is going to remove any screen tearing, but this can largely be avoided by compiling a real-time kernel which is easy to do since 6.12.

raw GPIO on rp5 allows for 9600Hz (20Mhz / (62*32 pixels)) refresh rate for 64x64 panels. With 32bits of color depth that should be 300fps. The panels themselves support 32Mhz operation, which should allow up to 488fps @32 bits of color depth.

ladyada, 2025-03-11:
@bitslip6 its limited by the PCIe interface. bitbang implementation sounds like an excellent project for you to take on!

bitslip6, 2025-03-11:
I was hoping that DMA to PIO would surpass 20Mhz.  Here is a multi threaded bitbang repo for hub75 oon pi5 with OpenGL support: https://github.com/bitslip6/rpi-gpu-hub75-matrix

marcmerlin, 2025-03-11:
@ladyada  thank you for the update and for the work on this, this is very exciting.
But to make sure I understand, is it indeed slower than bit banging on Pi4?
I can do 384x256 with a Pi4 and 3 channels (well, that was 2020, but things are the same today): https://marc.merlins.org/perso/arduino/post_2020-03-13_RGB-Panels_-from-192x80_-to-384x192_-to-384x256-and-maybe-not-much-beyond.html
Sure, I had to bring down color depth and use a few things to increase bit rate back to 100Hz or so, but it's 128x256 per channel, or 8 times more panels per channel.
it's not really 8 times more since my color depth was 21bits, but still.

So if I understand correctly, while DMA on rPi5 sounded exciting for bringing a stable output separate from the CPU, it now seems to be slow enough that it's not going to be a contender compared to bit banging?
That said, maybe bit banging on Pi5 will give a cleaner waveform that will allow outputs that work without gpio_slowdown because panels are unhappy with the waveform?

bitslip6, 2025-03-12:
@marcmerlin That seems to be the case. Pi5 GPIO is capable of 20Mhz with a good waveform, sounds like PIo is doing about 6.5Mhz and is limited at the PCIe interface (per ladyada). Linux 6.12 has support for the real-time kernel (you can also apply the RT patches which are backported to the 5.x kernels; there is a write-up on my github repo). Using a real-time kernel removes the flickering associated with kernel interruptions for pure GPIO. Not quite as good as hardware, and it does eat up 1 full core, but any flicker is not perceptible to humans. The main advantage of using PIO is lower CPU usage. Using PIO and OpenGL should almost eliminate CPU usage, but I don't see how this is an advantage for anything other than battery operation.

sub-zerohm, 2025-04-08:
I'll have to look into that real-time kernel.  I just set up a 256x192 display using this new PIO method on the pi 5 And it does suffer quite a bit from flickering when I run virtual display example on mednafen emulator like in tutorial.  It looks really good though, would be a great way to enjoy the retros in between the AAA releases

[10 more replies]