hzeller/rpi-rgb-led-matrix #1605: Working on Pi3, gibberish on Pi4

prosd, 2023-11-22:
Hello,

I have a chain of 4x 128x64 1/32S panels connected to a Pi3 (Model A+ with 512 Mb) via an Electrodragon Hat.
When I run the -D0 demo everything works fine (altough there is a little flicker)

This is the command I'm using:
`sudo ./examples-api-use/demo -D0 --led-cols=128 --led-rows=64 --led-slowdown-gpio=5 --led-chain=4 --led-parallel=1 --led-row-addr-type=3 --led-multiplexing=0
`
[image]

When I take out the SD card and put it into a Pi4 (tried 2 Model B+'s, one with 2Gb and one with 4Gb, with the same Electrodragon hat) and run the exact same command, the display is unrecognizable:  dots and horizontal stripes are shown where the rotating square is supposed to be. (See picture 2).

Picture 2 command: same as picture 1 command
[image]

Changing `slowdown-gpio` slightly improves the displaying. With a value of 2 I can - sort of- distinguish the rotating square, however the image is still very very malformed with a lot of horizontal stripes in the neigboorhoud (see picture 3).

Picture 3 command (best result on Pi4):
`sudo ./examples-api-use/demo -D0 --led-cols=128 --led-rows=64 --led-slowdown-gpio=5 --led-chain=4 --led-parallel=1 --led-row-addr-type=3 --led-multiplexing=0 --led-slowdown-gpio=2
`

[image]

I experimented with the various swithes like `--led-pwm-lsb-nanoseconds`, `--led-no-hardware-pulse` and `--led-slowdown-gpio` but I could not get the image as it is supposed to be.

Any suggestions on this would be greatly appreciated!

hzeller, 2023-11-22:
what do higher values of slowdown GPIO bring ? A Pi4 is usually so fast, that you might need a value around 5

prosd, 2023-11-22:
Thank you for your fast response!

I tested up to a value of 6 (modified the boundaries check in your code for this), but found that beyond 2 it gets worse again.
Suprising, because I had to use a value of 5 earlier on 64x32 indoor panels connected to the same Pi4.

HarikrishnanBalagopal, 2023-12-02:
> what do higher values of slowdown GPIO bring ? A Pi4 is usually so fast, that you might need a value around 5

Can you please call out this option in the README more prominently?
I have a Raspberry PI 3B+ rev1 with 1GB ram running Raspbian OS 64bit.
Connected to 4 64x64 panels in a single chain and the demos were working only for one panel. The other panels were showing trash.
Tried a bunch of options and then found this thread and used `--led-slowdown-gpio=5` to finally get the demo working on all 4 panels properly.

Ry-Kode, 2023-12-03:
@HarikrishnanBalagopal 
Thank you for your comment.

 I am aware of the --led-slowdown-gpio switch, and tried all combinations (1 up until 5), as mentioned in my previous comment:

> I tested up to a value of 6 (modified the boundaries check in your code for this), but found that beyond 2 it gets worse again. Suprising, because I had to use a value of 5 earlier on 64x32 indoor panels connected to the same Pi4.

However, I was unable to get good results with any of the --led-slowdown-gpio settings

Ry-Kode, 2023-12-05:
So far no progress, regardless of which software adjustments I make. 

I don't know if the electronics used on my panels can be the cause of this issue? Is there somone out there who is more experienced in this matter who can comment on this? 

[image]

davemaster, 2023-12-05:
check your wiring... again, and again

davemaster, 2023-12-05:
> @HarikrishnanBalagopal Thank you for your comment.
>
> I am aware of the --led-slowdown-gpio switch, and tried all combinations (1 up until 5), as mentioned in my previous comment:
>
>
> However, I was [...]

What was your last test? Picture

and picture your wiring, and check it again, and again, wire by wire (connectivity, missing lead, etc)

Ry-Kode, 2023-12-06:
@davemaster, thank you for the suggestion. 
I did double and triple check my wiring. 
Also tried with all new components: new psu, other panel, other Electrodragon hat (tried V1 and V2) and other Pi: exact same result. Meaning: working fine with a Pi 3, distorted image on Pi 4.

As said, when I replace the panels with 'regular' P5 64x32px everything works perfectly, on Pi3 AND Pi4.

If this can be helpful, this is a picture of the Pi4 with slowdown-gpio set to 4:

[image]

ledvinap, 2023-12-06:
@Ry-Kode : You can try chaining working panel before problematic one. It may reveal some signal integrity issues.
Also, try even higher led-slowdown-gpio (recompile may be necessary) or look at my PRs / my repository clone. I did experiment with a bit different approach to RPI timing