hzeller/rpi-rgb-led-matrix #1609: The left led copies the right end on panel 104x52

amrikhsn, 2023-12-06:
please someone help me.. i am using p3 104x52 one panel. with direct regular wiring on pi 4. but i have problem. first row to 8 leds to 52 copy the last led.. 

https://github.com/hzeller/rpi-rgb-led-matrix/assets/142022228/4114c4fe-6002-4e28-9c3e-7070b3e247f5

amrikhsn, 2023-12-07:
pleaseee helpppp

hzeller, 2023-12-07:
You are not providing enough information about your issue.

The cryptic title and description sound like some sort of symmetry issue, but not really explaining what you see vs. what you expected. Then you show a video of something that is supposed to be symmetric anyway, so that doesn't help in discerning what you mean.

Also, you didn't provide any command line flags you were using. How did you do the wiring, manually (with or without a level shifter) or with an adapter board. Where did you get the unusual 104x52 board from etc. All kinds of information anyone would need to figure out what is going on.

amrikhsn, 2023-12-08:
I tried `--led-cols=112` but still had problems. he doesn't duplicate the last part but duplicates the side if I'm not mistaken

https://github.com/hzeller/rpi-rgb-led-matrix/assets/142022228/63612a4d-45ca-4e41-8c46-7ee26f4fc82e

amrikhsn, 2023-12-08:
I tried `--led-multiplexing=3` initial leds don't duplicate the final leds... I don't understand whether this is a software or hardware error

https://github.com/hzeller/rpi-rgb-led-matrix/assets/142022228/4291ceb1-ca37-40f1-931c-894f7ae5c11b

ledvinap, 2023-12-08:
IT seems that the shift register is 16bits long, but only second half is connected to physical leds. With --led-cols=112 you get fresh frame data into leds, but there is 8-led gap that is not displayed.
You can write new multiplexer for this panel. Or modify the program generating data so that it does generate the gap internally

andytheengineerguy, 2023-12-10:
Hi @amrikhsn, I haven't experienced this exact problem before, but I have brought up 104x52 res panels so here are some suggestions that may help

1. Use the [Electrodragon shield](https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/). It's super cheap, creates a robust connection (no jumper wires), and allows you to use ribbon cables. Plus, always a good idea to have a buffer for high speed IO.
2. Make sure you've accounted for if the panel is HUB75 or HUB75E. If it is 'E', you may need to wire it up slightly differently than a normal HUB75E. For example, with the Electrodragon shield, you needed to add two jumper wires to the PCB (process explained [here](https://w.electrodragon.com/w/RPI_RMP_Guide#P4_64x32_.28tested.29)). 
3. For my 104x52 panel, you might have a mapping problem. I would be surprised if if worked out of the box with led-multiplexing = 0 because I needed to write a custom mapping for mine. This depends on things like the scan rate (aka how much multiplexing is going on), the manufacturer, etc. But this is almost impossible to figure out, because, the panels (especially if they're from Aliexpress) typically have 0 documentation. Anyway. For my panel, it needed a new mapping that was very similar to the ZStripeMultiplexMapper, but with a tile height of 13. I edited this mapper so that you can pass in _width_ and _height_ as variables, as shown:

[image]

And then when I added the mapper into the list, I defined the width and height variables as 8 and 13 respectively:

[image]

Good luck, hope something here get the panels working, or leads you down the right path. Keep us updated!

amrikhsn, 2023-12-19:
Thank you @andytheengineerguy for helping. but I'm running out of ideas to fix it. I was thinking about buying an electro dragon board but I couldn't wait for it to arrive. so I decided to buy a new p4 64x32 1:32S board [image]... and still use regular/direct wiring mapping. and I get a problem.. when I use the command `sudo ./demo -D0 --led-no-hardware-pulse --led-cols=64 --led-rows=32` the result is like this [image]
and if use different animation `sudo ./demo -D0 --led-no-hardware-pulse --led-cols=64 --led-rows=32`  -D 1 runtext.ppm
[image]
So the problem is that point 1 in the led row starts at row 16.. 

if i use different pixel with row 64. `sudo ./demo -D0 --led-no-hardware-pulse --led-cols=64 --led-rows=64` and `sudo ./demo -D0 --led-no-hardware-pulse --led-cols=64 --led-rows=64  -D 1 runtext.ppm` . the result
[image]
[image]
when I use row 64 configuration it runs fine. Row 1 is in the right position. but we have a symmetry problem. because I use one panel... 

I tried configuring --led-multiplexing from 1 to 18 but the results did not match. I have also tried --led-row-addr-type from one to 4 but still no results. so does --led-slowdown-gpio etc. 

then I tried to create a custommapper in multiplex-mapper.cc with the help of chatGPT, and add this is the code
```
class CustomMapper : public MultiplexMapperBase {
public:
    CustomMapper() : MultiplexMapperBase("CustomMapper", 1) {}

    void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const {
        // Assuming panel_rows_ is the number of rows in a single panel
        // Adjust the mapping calculations for a 64x32 panel with a 1:32 scan rate

        int shift_distance = 16;  // No shift needed
        const int panel_width = 64;
        const int panel_height = 32;

        // Calculate the adjusted y-coordinate based on the shift distance
        int adjusted_y = (y - shift_distance + panel_height) % panel_height;

        // Map the adjusted x-coordinate and y-coordinate
        *matrix_x = x;
        *matrix_y = adjusted_y;
    }
};
```

i try using  `sudo ./demo -D0 --led-no-hardware-pulse --led-cols=64 --led-rows=32` and `sudo ./demo -D0 --led-no-hardware-pulse --led-cols=64 --led-rows=32` and the result 
[image]
[image]

rows 1 to 16 may have been correctly placed in the correct place. but he didn't print anything..

I ask for his help. because I don't understand the code. I hope a great library maker @hzeller can help me 😊

andytheengineerguy, 2023-12-20:
Hi @amrikhsn, good on you for following it to this extent, you've clearly put a bunch of time into this!!

This is reaching the limits of my knowledge of this library tbh - I've not seen these types of problems you're describing. This might be a question for the almighty hzeller if he has time to respond 😂👌

But in lieu of that:

1. Again I recommend just ordering some electrodragon shields and trying them when they arrive. You may as well, they are like $3 and you're running out of other options anyway
2. Try explicitly setting the variables for parallel and series panels. i.e, for your single panel, `--led-chain=1` and `--led-parallel=1`. Generally speaking, if the library thinks that there are 2 panels in parallel, then it can lead to the behavior you're seeing where only the top half of the image is displayed.
3. Make sure you have accounted for the HUB75 vs HUB75E point I mentioned above.

amrikhsn, 2023-12-20:
ha ha. I don't understand the code flow. I asked chatgpt until I was confused, he answered.. I want to buy Electrodragon but the cheapest delivery package is estimated to take up to one month.. I have also tried `--led-chain=1 and --led-parallel=1`.. or also like the configuration `--led-cols=32 --led-rows=16 --led-chain=2 and --led-parallel=2`. the results are still the same... but I'm sure this time Mr. @hzeller has the solution 😅

xuniuer, 2023-12-27:
As known, only one panel with 64x32 size, just to use `--led-row=32 --led-col=64 (optional: --led-chain=1 --led-parallel=1)` is enough.