hzeller/rpi-rgb-led-matrix #1611: How would i set this up with a cron schedule?

gguy825, 2023-12-21:
How would you set thip on a crontab exactly? Ive had no luck doing this. The command im running is 
`sudo ./led-image-viewer -f -s -w600 -C Images\ Panel/*.png --led-cols=64 --led-rows=64 --led-chain=2 --led-gpio-mapping=adafruit-hat-pwm --led-slowdown-gpio=4 --led-brightness=50`
I put it in a .sh file also but still didn't run properly.

jaygarcia, 2023-12-30:
Cron is like 30+ years, loads of examples, tutorials online. https://crontab.guru/examples.html

You can't use sudo because it's an interactive program that requires password to escalate privileges, so you'll need to put it in root's crontab.

An alternative is to setup a service that runs the program at set time intervals using escalated privileges as well. 
https://medium.com/@benmorel/creating-a-linux-service-with-systemd-611b5c8b91d6

p40lop, 2024-01-09:
I believe I have a similar issue. In my case I directly edited /etc/crontab to run a .sh script which contains the command to display my pictures.
In the command I have used only absolute paths. 
I am absolutely not an expert in this kind of things, but I ran some tests to figure out what could be the problem.
In my case, everything works fine with a single image, but it doesn't work with GIFs or with multiple images, neither selected with * or manually inserted.
I also tested the demo D0 (the spinning cube) and that works.
I suspected that the loading time could be an issue so I tested with different image sizes: an image of about 144kB works with a little bit of flickering, one of about 1MB does not work at all.
In all these cases the script runs fine from the command line.
I tried googling if Cron limits does limit time to execute a task but for now I found nothing.

jaygarcia, 2024-01-09:
Could it be that your paths are not correct? Loading time for a program is not an issue for Cron.

p40lop, 2024-01-11:
I believe my paths are correct, as both running the same script from the command line and running the script from cron but loading the lighter image (which is in the same folder) work.

p40lop, 2024-01-11:
I found a workaround: I first output whatever pictures I want to show to a stream file using the -O flag and then I use it in the script I run with cron.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin