hzeller/rpi-rgb-led-matrix #1618: HUB320 support

davidkessler-ch, 2024-01-18:
So all the matrices listed use HUB75, but for bigger matrices there is the similar HUB320. Are there plans to support that?

I have a 256*128 pixel matrix with HUB320, a 1/64S see [here](https://de.aliexpress.com/item/1005004995908838.html)

wata-net, 2024-01-20:
There is nothing the library can do about it.
All you have to do is make a conversion cable.
[image]

Keniis0712, 2024-03-06:
> There is nothing the library can do about it. All you have to do is make a conversion cable. ![1](https://private-user-images.githubusercontent.com/122246369/298236731-24f4ba41-b7f4-4bdc-9480-b55c26eba86e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDk3MzEyODksIm5iZiI6MTcwOTczMDk4OSwicGF0aCI6Ii8xMjIyNDYzNjkvMjk4MjM2NzMxLTI0ZjRiYTQxLWI3ZjQtNGJkYy05NDgwLWI1NWMyNmViYTg2ZS5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzA2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMwNlQxMzE2MjlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hYTMzZGNlMTM4NTJlZTI1OTk0ODU1MTQwNjI0MGNlNWRhNzZhYTUzMGM3ZTE4M2UzYTczMjBiZjE1ODNlZWZlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.8Zz_IzX-DetGVMCYLT0NXJ4G82detOyBw3mQ7SMSM-Q) However, most HUB320 LED panels use driver chips with built-in SRAM. This library does not support driver chips with built-in SRAM. (This is not related to the HUB320 issue.)

I have a special driver chips withnot built-in SRAM and I only find the library for HUB75. How can I use this kind of library to control a HUB320 screen?

krambs, 2024-08-25:
Anyone able to get this library working with HUB320 and 256x128 yet? Thanks!

marcmerlin, 2024-11-09:
@krambs  it's a new setup, new chip, new way to drive it and new cable.  Unless you write it yourself or pay someone to write it, it's probably not likely to happen unfortunately.
Also, note that panels that do not have an output plug, cannot be chained, so you'll really want an FPGA card with lots of output cables to run a bunch of those.

sub-zerohm, 2024-12-28:
I've looked into this myself, if you lookup Kevtris on YT he has a video of how he reverse engineered one to work with jumpers soldered the NT Mini's NES cartridge pins (with a cart adapter as to not ruin the actual NT) and the open FPGA with his own custom gameboy core....but those are 26-pin connectors, and the pi only has 40 pins.  20 of which are actually not ground pins, but that means you need 21 pins to get one panel working, leaving no possibility for more than a single panel since they're designed to not be chained.

I want to port the library to an NES emulator but I don't think the pi4 can handle it, waiting patiently for the library to come to the pi5 🙌 hzeller is the man

sub-zerohm, 2024-12-28:
Actually was wondering if these have sram.  Can't tell by looking.  I ordered them direct from the supplier

[image]

rxchit, 2025-02-08:
I'm also working with one of the panels with HUB320 connectors but the pinout is slightly different, can you guys point in any direction to get it working with pi4/5[image]

marcmerlin, 2025-11-07:
@martodg  you're supposed to make a new issue, not to post in a random unrelated one.
But there aren't posts indeed because those panels aren't supported as per the README.  
As of right now, I have not seen a single 128 line panel work with the existing libs, and it's likely because without using PWM the refresh rate would be terrible, and PWM chips are completely different, and unsupported, until someone writes a driver for one of them. No one has offered to do so as of today.

sonycman, 2025-11-08:
@martodg 
Hi. Successfully running similar panel (DP3364S + RUL5258 + HUB320) on custom hardware (FPGA).
Now just received same panel as yours - DP3364S + RUL5158 + HUB75E.
Seems almost the same both of them, except less data channels on latter, thus it might need higher clock or dual data rate for DP3364...

martodg, 2025-11-08:
@sonycman 
I have created a new issue:
https://github.com/hzeller/rpi-rgb-led-matrix/issues/1821
If you could post on that issue and assist, that would be great. Thanks again