hzeller/rpi-rgb-led-matrix #1624: Issue with Displaying Pixel Propperly

glappa, 2024-02-20:
Hello, 
Im using this Matrix Pannel :

glappa, 2024-02-20:
i also tried it without the "--led-panel-type=S3800467M05334" Parameter and the Matrix still just shows a few Pixels instead of the Demo

Nam0, 2024-03-21:
Could you send a picture of the back of your LED Panel?

davemaster, 2024-03-21:
Greetings, 

You wanna be helped? First, take many pictures to your panel, specially to: HUB-75 LED INPUT, code of the RGB PANEL.
Second, use the command :

Command used on the Raspberry PI 4
"sudo ./demo -D 7 --led-rows=64 --led-cols=64 --led-gpio-mapping=adafruit-hat --led-slowdown-gpio=5"

and record a video of the result. 

Best regards

glappa, 2024-03-21:
Hello @davemaster.
thank you for your reply!
here are some picture of the back of my matrix!
[image]
[image]
[image]
[image]
[image]
[image]
[image]

as well as a video with the command you provided!

https://github.com/hzeller/rpi-rgb-led-matrix/assets/119673858/ec809e73-d953-4486-85d2-882faef5d0d8

https://github.com/hzeller/rpi-rgb-led-matrix/assets/119673858/edf3e228-e31c-4a25-ba5b-3875730ac5c7

hope this can help!

Nam0, 2024-03-23:
Hi Glappa,
Thanks for all of the pictures and videos, it looks like it might be an issue with how your E Line is set up, I was running a similar setup, slightly different board but same Hub 75 configuration. Could you try and solder a jumper from Pin 4 (OE) to Pin 18 on the RPI Bonnet board and run it again? If you want to double check before soldering it you should be able to steadily hold a wire on the two contacts and it should get reflected instantly on the board.

glappa, 2024-03-24:
Hello @Nam0,
Did you mean it like this?
[image]

[image]

[image]

Same Result:
https://github.com/hzeller/rpi-rgb-led-matrix/assets/119673858/fb277d97-d044-4857-a07c-1e36440d6b20

Nam0, 2024-03-24:
Gotcha, and just to double check you have the E and 8 pins on the underside of the bonnet board bridged correct?

glappa, 2024-03-24:
this is the backside of the Bonnet:
[image]

[image]

[image]

Nam0, 2024-03-24:
Ah I see you don't have the E line pin soldered up, you will need to go thru and bridge the connector from E to 8 for these two pins
[image]

glappa, 2024-03-24:
like this @Nam0 ?
[image]

glappa, 2024-03-24:
Thank you very much @Nam0 IT WORKS!
https://github.com/hzeller/rpi-rgb-led-matrix/assets/119673858/3f06e4a6-fef0-4d0d-955b-ebf8f63dc98a

glappa, 2024-03-24:
marked as closed

closed 2024-03-24 by glappa