hzeller/rpi-rgb-led-matrix #1627: I don't like gpio_slowdown

targettadams, 2024-02-29:
This way of introducing a delay between writing to the GPIO bits looks hacky. With some recent panels I had to reduce the slowdown to 6 to get them to work (pi4). Why don't we just use an explicit, tunable time delay using the relevant pulser? Or am I missing something.

ledvinap, 2024-02-29:
you can look at https://github.com/hzeller/rpi-rgb-led-matrix/pull/1479, 

This part is writing parallel data to GPIO, pulser won't help here (or it will be far too slow). Port access is done through AMBA bus, it does queue write requests and GPIO is processing them in chunks (at over 100MHZ on pI4)

targettadams, 2024-02-29:
Interesting. Ghosting is my issue currently. I can only get rid of it by suppressing the refresh rate to an unacceptably low level. I'll look at your branch. I may have some questions if that's ok. So - you were able to get good refresh rates with no ghosting at all? And the brightness was not inhibited at all?

targettadams, 2024-02-29:
Unfortunately, it didn't change my ghosting issue, but it is a lot nicer solution to some arbitrary parameter. Good work!

targettadams, 2024-02-29:
Yes, I set gpio_slowdown = -1. It works fine and, in my opinion, is much better than using the usual approach.

Yes, I suspect the ghosting is something else. I use an adafruit bonnet, so I am not using the PWM pulser. I will try the solder hack next, Hard to tell, may just be the panels I'm using. Every panel I've use has its different quirks.

targettadams, 2024-02-29:
Tbh the ghosting is not major. But once you notice it, it's difficult not to become perfectionist

ledvinap, 2024-02-29:
> gpio_slowdown = -1. It works fine and, in my opinion, is much better than using the usual approach.

It's also less that half of refresh rate compared to reliable multiple write approach. And fraction of actual panel speed.

bluelasers, 2025-04-07:
@targettadams There are a couple things to ghosting. If you wanna chat start one here: https://github.com/bluelasers/rpi-rgb-led-matrix/discussions

bluelasers, 2025-04-07:
@ledvinap I tend to agree with -1 as the way forward. Thank you for adding that! Ghosting is caused by software currently. Most cannot benefit from the corrected software. I will not say much here due to the bandwidth of a conductor.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin