hzeller/rpi-rgb-led-matrix #1628: Demos work, utils dont

emersonsc, 2024-03-03:
Raspberry Pi Zero 2 W
(2) 16x32 Adafruit Matrix Panels
Adafruit RGB Matrix Bonnet
Raspbian Bullseye 32bit Lite

Everything installed, no errors. 

Example: 
`sudo ./text-scroller -f ../fonts/9x18.bdf -B0,0,255 -O0,0,100 -C255,0,0 --led-rows=16 --led-cols=32 --led-chain=2 "Contrast outline"
`
typical "CTRL-C for exit." output, but nothing on the Matrixs.

` sudo ./demo -D4 --led-rows=16 --led-cols=32 --led-chain=2
`
produces entire matrixs to fade in and out with different colors (as it should).

utemorley2012, 2024-09-24:
I am having the same issue.

hzeller, 2024-09-24:
If you use the Adafruit hat/bonnet, you need to give the option `--led-gpio-mapping=adafruit-hat` (but if you just use the basic hat I also strongly suggest to do a little hardware mod https://github.com/hzeller/rpi-rgb-led-matrix?tab=readme-ov-file#improving-flicker )

The demos should also not work without this flag, but I suspect you actually did not use this library here, but the Adafruit fork, where they patched it in in some places ?

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828