hzeller/rpi-rgb-led-matrix #1635: Support for 160x80 panel

dev-markus, 2024-03-21:
Does this library support a panel with a resolution of 160x80 pixels? When I run the demo, it displays a message indicating a limitation regarding the number of pixels.

Product: https://de.aliexpress.com/item/1005005326933468.html?spm=a2g0o.order_list.order_list_main.15.17fe5c5fjr60qt&gatewayAdapt=glo2deu

krux77, 2024-03-23:
Hi Markus,

I am also trying to get a 160x80 pixel panel to work.

Mine has SM16380sc driver chips (supposedly FM6363 clone).

https://a.aliexpress.com/_EwMIwEv

Which hat are you using?

dev-markus, 2024-03-23:
HAT: Adafruit RGB Matrix Bonnet for Raspberry Pi
Link: https://www.amazon.de/dp/B07DNBLXV2?psc=1&ref=ppx_yo2ov_dt_b_product_details

krux77, 2024-03-23:
i have that one too, 
what chips on your panel?

iojan-iot, 2025-07-10:
Dear ones, I also bought a P2-160x80 panel on Aliexpress that was very strange, it has, like most of it, a HUB75E port, which implies that it has lines A, B, C, D and E, clearly you can deduce that by looking at screen printing.

But the strange thing is that line D and E are connected to GND, measured with a digital tester, it seems that lines D and E are not used.

To finish complicating it, use the FM6363C chip, which is why I will arrive at this forum.

Before I throw it away, I ask, does anyone have any recommendations or help?
Thank you very much now.

closed 2025-11-27 by marcmerlin