hzeller/rpi-rgb-led-matrix #1636: How can i display costum Content on my Matrix

glappa, 2024-03-24:
i've seen Youtube videos where they display GIF's and so on.
but i can't find how to do so.
can anyone help me out?

VitekOpach, 2024-03-25:
You can try commands in utils such as:
`sudo ./led-image-viewer animated.gif `

Check out all the commands [here](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils)

glappa, 2024-03-25:
is there a way for me to show the live crypto prices?

VitekOpach, 2024-03-26:
You have to write a code for that one. Read the [repository](https://github.com/hzeller/rpi-rgb-led-matrix) **carefully**, everything is written there.

VitekOpach, 2024-03-26:
> is there a way for me to show the live crypto prices?
> 
> or have random content be shown on the Matrix, like other pricey LED products do it?

You have to write a code for that one. Read the [repository](https://github.com/hzeller/rpi-rgb-led-matrix) carefully, everything is written there.

closed 2025-11-27 by marcmerlin