hzeller/rpi-rgb-led-matrix #1640: 64x64 LED panel 1/16 scan requires new multiplex-mappers

martodg, 2024-04-05:
Hi,

I recently purchased a P2.5 64x64 outdoor LED Panel https://www.aliexpress.com/item/1005005908894934.html
and the led-multiplexing doesn't seem to work for the 1/16 Scan.

I have an indoor P3 64x64 1/32 Scan panel that works fine.

How can I implement a 19th custom multiplex-mappers.cc file to the rpi-rgb-led-matrix?

Thanks

romsera, 2024-04-11:
Hi,

I'm in the same situation, is there anyone who could guide us in creating a new mapping?

Thx a lot.

Romain.

jwee1369, 2024-05-23:
Same, any support for this?

davemaster, 2024-05-23:
Greetings,

Try to start analyzing your panel, using an step, by step pixel drawing, one by one so your see on your panel the pattern. I remember one code in Arduino, x - column, y  - row, using two for loop.... with a delay, and voila, then, put your new pattern into your [multiplex-mappers.cc](lib/multiplex-mappers.cc); if success, share here please.

davemaster, 2024-05-23:
Greetings,

Try to start analyzing your panel, using an step, by step pixel drawing, one by one so your see on your panel the pattern. I remember one code in Arduino, x - column, y  - row, using two for loop.... with a delay, and voila, then, put your new pattern into your [multiplex-mappers.cc](lib/multiplex-mappers.cc); if success, share here please.

[here in this link ](https://github.com/hzeller/rpi-rgb-led-matrix/issues/592)from this repository, a way to understand better what you need to do... I'm out of time right now.

davemaster, 2024-05-23:
> Hi,
>
> I recently purchased a P2.5 64x64 outdoor LED Panel https://www.aliexpress.com/item/1005005908894934.html and the led-multiplexing doesn't seem to work for the 1/16 Scan.
>
> I have an indoor P3 [...]

better that the link you get your panel, pictures of it, of its ICs, its hub75, any code or serial... Thanks in advance

davemaster, 2024-05-24:
here, my 64x64 32S P3 Screen, I had to mod the Electrodragon rpi rgb adapter, FIRST, look where in your panel HUB75 is the E pin... in my case, was in PIN 8. 

ALL details of my mod, [here in my video ](https://www.youtube.com/watch?v=sBRi85QoGuY)

martodg, 2024-05-24:
Hi All,
Thanks for the posts.

Turns out the fix for the LED module was the --led-row-addr-type=3 Flag. Once I added this flag I was able to use Python and send images and text to the board. I've since created a 64 * 128 python script that generates QR Codes and Images based upon 4 GPIO inputs. It's a demo I'm creating for car park access. The demo can also output network / time settings. Have a look at the Video (compressed)

https://github.com/hzeller/rpi-rgb-led-matrix/assets/166124868/914e8292-da0d-4d04-8df0-d3f97b3435f1

Regards,

Martin

ixcue, 2024-07-15:
but I can’t connect my 1/16 matrix using the adafruit bonet board. no one knows how to help me?[image]
[image]

martodg, 2024-07-15:
Hi ixcue,

I noticed you have --led-row-addr-type=0 flag, with --led-multiplexing=4

Have you tried --led-row-addr-type=3?

ixcue, 2024-07-16:
> I noticed you have --led-row-addr-type=0 flag, with --led-multiplexing=4
> 
> Have you tried --led-row-addr-type=3?

Unfortunately, this didn’t help, I left one panel and tried all the modes...

ixcue, 2024-07-16:
Video https://photos.app.goo.gl/kLfGcouEHaDJKncq9

ixcue, 2024-07-16:
sudo examples-api-use/demo --led-rows=64 --led-cols=64 -D 3 --led-gpio-mapping=adafruit-hat --led-slowdown-gpio=2 --led-multiplexing=0

[image]

closed 2025-11-27 by marcmerlin