hzeller/rpi-rgb-led-matrix #1642: Chain 4 RGB LED Matrix (64x32) for display

mandarp22, 2024-04-10:
Hello Everyone,
I am trying to chain 4 RGB LED Matrix displays in attached format to create a display of 128x32.
I want to display big text or numbers on display as part of my project , so I need to use these 4 displays to make one giant display.
I have Adafruit HAT and Raspberry PI4

I tried to use several samplebase.py options but not able to figure out on how to make them chain to use as single display.

Can you please help?
[image]
[image]
[image]

davemaster, 2024-04-10:
[image]
Greetings. This is the way I get worked, but with 3x2 arrangement. U must use the U-MAPPER, I don't know if that feature is still used in the latest version of rpi-rgb-led-matrix, but You must have Rpi 3-Rpi4 for this.

./demo --led-chain=4 --led-pixel-mapper="U-mapper"

in this link, an example -> [example u-mapper](https://github.com/hzeller/rpi-rgb-led-matrix/tree/a3eea997a9254b83ab2de97ae80d83588f696387/examples-api-use#remapping-coordinates)

mandarp22, 2024-04-10:
Thank you Dave. That worked perfectly.
Thank you for your help and response

davemaster, 2024-04-10:
You're Welcome, I'm glad your problem is gone... God bless You

mandarp22, 2024-04-14:
[image]

Thanks Again @davemaster

mandarp22, 2024-05-28:
I got new outdoor LED's and I am using same setting but I am getting output as below

./demo -D0 --led-slowdown-gpio=4 --led-chain=4 --led-pixel-mapper="U-mapper"

https://github.com/hzeller/rpi-rgb-led-matrix/assets/166573021/10310a4a-be0f-4b49-b0b2-02a02adff16c

mandarp22, 2024-05-28:
I am using same setup 4 -   64 x32 P5 LED Outdoor panels
[image]
[image]
[image]
[image]

mandarp22, 2024-05-28:
Old Indoor LEDs were fine with code, as picture above I was able to display score correctly.

However with same setup for these LEDs. I am getting below.

I also tried setup changing row=64 and cols=128 , but still same

[image]

mandarp22, 2024-05-28:
[image]

mandarp22, 2024-05-28:
[image]

mandarp22, 2024-05-28:
Thank you for help in advance

mandarp22, 2024-05-28:
actually i was able to fix issue by setting --led-multiplexing =1

closed 2024-05-28 by mandarp22

davemaster, 2024-05-29:
> actually i was able to fix issue by setting --led-multiplexing =1

Share Your success with a picture, better with a Video. Best regards