hzeller/rpi-rgb-led-matrix #1643: 64x32 Display half off: 16 vs 32 scan mode conflict?

acastles91, 2024-04-10:
Hello,

I recently purchased 5 of these modules from ebay ([https://www.ebay.de/itm/285579711444](https://www.ebay.de/itm/285579711444)) and I can only get one half of the display to light up when I run the demos. I know it's not a hardware problem because I the issue is persistent with 5 different modules. This is the second screen that I put up using this  amazing library for an art project, and previously it worked, although with different more expensive modules. When compared to the previous modules, the difference I see is that those were 16S, whilst the new ones are 32S. As written in the back of the modules, they are:: 64x32, P4 2121, 32S, V2.0. What does 16S and 32S refer to exactly? Can I pass this value as an option to the demos?

acastles91, 2024-04-11:
> Greetings,
> 
> Could You share the command are You using? Also an Image of your rgb panel output
> 
> PD- First thing I always do, check connection, check correct mapping for that connection (regular, adafruit-hat, etc) then, power supply 4A (for each panel)

Thank you for your help. This is the command I'm using:
'''
demo -D0 --led-rows=32 --led-cols=64
'''

And this is how it looks. I'm attaching also a picture of how it looks from behind, and I'm supplying 5V with a bench power supply.
[image]
[image]

davemaster, 2024-04-11:
try this:

demo -D0 --led-rows=32 --led-cols=32 --led-chain=2 --led-multiplexing=1 (then 2, 3, .... etc), and picture

best regards

PD. What RPi are You using?

Nolicx, 2024-08-10:
Hey there, i have the exact same problem.
I also bought the panels from the same vendor, but about a year ago.
i've tried setting cols as 64 and cols as 64 with led-chain=2, didnt change anything.
I've also tried all multiplexing and all row-addr-types using:
```
for MULTIPLEXING in {0..17}
do
    for ADDR_TYPE in {0..4}
    do
        echo "Running demo with Multiplexing=$MULTIPLEXING and Addr-Type=$ADDR_TYPE"
        sudo ./demo -D0 --led-rows=32 --led-cols=32 --led-chain=2 \
        --led-multiplexing=$MULTIPLEXING --led-row-addr-type=$ADDR_TYPE --led-no-hardware-pulse
    done
done
```
However here the standard-values have the best result, which is only using the lower half.

Which i found interesting was: When Using the E-Connection, only the top half of the matrix is used, without only the bottom half is used.

I've tried using a Rpi 3 and a Rpi Zero W, same output on both.

Here are some Images:
The Matrix, that i'm trying to use:
[image]

And some Multiplexing-Adressing-Combinations:
[image]
[image]
[image]

So yeah, i'm at a loss and would appreciate every help.

davemaster, 2024-08-10:
> Hey there, i have the exact same problem. I also bought the panels from the same vendor, but about a year ago. i've tried setting cols as 64 and cols as 64 with led-chain=2, didnt change anything. [...]

According your picture, your RGB Panel is a 64x32 32S panel. First STEP, how are You wiring this.... Details please.... close picture, better.

Nolicx, 2024-08-10:
Hey there,
first of all thank you for your fast response.

So the panel itself is called "P4-64x32-32S-2121-V2.0" and it uses, what to me seems like, a HUB75E Adapter.
[image]

I used the normal wiring guide without any bonnets or hats.
[image]
[image]
[image]

And I am currently using a RPi3, however Zero W has the same results.

Basic command i used:
 `sudo ./demo -D0 --led-rows=32 --led-cols=64`

I also tried using different combinations with the D and E Pins:
D and E connected:
[image]

Only D connected:
[image]

Only E connected:
[image]

D and E both not connected:
[image]

D and E switched:
[image]

davemaster, 2024-08-10:
Looks like a row problem... one wired cable can be break; I suggest You change all rows (A B C D E) wires, test EACH of them before you plug in GPIO and HUB75; Also, what POWER SUPPLY do You have, You need at least for this panel a 5A at minimum

Nolicx, 2024-08-10:
So i tried switching the A B C D and E wires and checked them with a multimeter - nothing changed.
I am using a generic 10A PSU to power the Pi and the matrix:
[image]
[image]

Nolicx, 2024-08-11:
@acastles91 Did you ever manage to fix this issue?

acastles91, 2024-08-11:
@Nolicx In the end my problem was that I was mixing different panels that looked exactly the same. So if you are daisy chaining several panels make sure that they are all either 32s or 16s and try which multiplexing works, I don't remember exactly what command I used.

Nolicx, 2024-08-11:
@acastles91 okay thank you.
Sadly I am currently only trying to address the single panel and the different multiplexing modes do not work.
So if anyone has another idea I'd be happy to try it out.

davemaster, 2024-08-11:
> So i tried switching the A B C D and E wires and checked them with a multimeter - nothing changed. I am using a generic 10A PSU to power the Pi and the matrix: [...]

THIS, only for 1 or two (max) panels... These panels need 5-7A at full bright with white color... and Wires (A,B,C,D,E) CAN'T BE SWITCHED, When I'd advice You to check those, I mean if wires are in good condition (sometimes, wires are broken)  YOU MUST strictly follow the wiring pinout. and CHECK that E pin, twice... 

I think your panel can be broken from vendor... or, the GPIO pins of raspberry ALSO can be break

Nolicx, 2024-08-12:
Hey there,
I know that you meant checking the wires, that's why I said I used a multimeter, to check if the cables can close a circuit.
I also only switched the D and E pin once to try it out, that's when I showed you in the comment.

I have already tried to rewire everything multiple times and I have already tried two separate pies and two separate panels (I actually have 2 of the exact same, but am only trying to control one at the moment).
And I cannot imagine having two broken pis (3 and zero w) AND two broken panels. As I said I rewired everything multiple times with new and checked cables.
But we will see, maybe I did do an error multiple times.

I will try to wire it once again with a pi 4 and another panel.
I will let you know, how it went.

Nolicx, 2024-08-12:
Okay so i switched to a RPi 4 and rewired everything and i still get the same results:
[image]
Should I open another Issue?

davemaster, 2024-08-13:
Greetings,

Only trying to help, this library works fine with 16S 32x32 panels (example, these from sparkfun [https://www.sparkfun.com/products/14646](https://www.sparkfun.com/products/14646) 

1. I have x7 of them, x6 of those from sparkfun in a[ 96x64 screen](https://youtu.be/DsmSmaUgFRw), and works perfectly with a RPi 3, and RPi 4, not RPi 5 compatibility yet. and NO --led-multiplexing parameter necessary for these panels, because they're compatible by default.

command used in my RPi 3:

sudo ./demo --led-rows=32 --led-cols=32 --led-chain=3 --led-parallel=2 --led-chain=1 --led-slowdown-gpio=2 --led-no-hardware-pulse -D 0

2. When I bouth I bought some 16x32 4S P10 outdoor panels, for a [Scoreboard](https://lh3.googleusercontent.com/33guE573LdqmPaoVta4Rh_QMzDe_EopMbU6ijtYCZ3oujrRO3NUl1JqwATNG_EFvLjw65A=w16383), I NEED to use --led-multiplexing=8, so the library generates the right flow of data to show info into panels correctly. If You don't have 16S panels from sparkfun or SIMILAR/Compatible, you must try the existent values for --led-multipexing, I think now, are around 18 different types of panels o ways to show info into panels supported... try again, one by one...

3. Some [experience I have](https://youtu.be/sBRi85QoGuY), many NIGHTMARES making this excellent library works with different kind of panels... 

Good luck, with patience, You will find the problem

Nolicx, 2024-08-13:
Hey,
thank you again for your help.
As i have stated befiore, i have already tried every possible combination of multiplexing and address types using
```
for MULTIPLEXING in {0..20}
do
    for ADDR_TYPE in {0..4}
    do
        echo "Running demo with Multiplexing=$MULTIPLEXING and Addr-Type=$ADDR_TYPE"
        sudo ./demo -D0 --led-rows=32 --led-cols=32 --led-chain=2 \
        --led-multiplexing=$MULTIPLEXING --led-row-addr-type=$ADDR_TYPE --led-no-hardware-pulse
    done
done
```
Maybe there needs to be another multiplexing method implemented to properly use this panel.
If i create another issue, will i get some mote help/guidance, or am i more or less on my own with this?

davemaster, 2024-08-13:
well, python.... 

but, where is the output of all these "--led-multipexing" ¿¿¿???

Advice, ENTER one by one each multiplexing option... come on, are not too much, only 17 as far I know...

1. sudo ./demo -D0 --led-rows=32 **--led-cols=64** --led-chain=2 \
        --led-multiplexing=$MULTIPLEXING  --led-no-hardware-pulse
        
 WHY --led-row-addr-type=$ADDR_TYPE ???? 
 
_this is where use it: "... The 64x64 matrixes typically come in two kinds: with 5 address lines (A, B, C, D, E), or (A, B); the latter needs a --led-row-addr-type=1 ....·_

I no doubt You're doing right, the wiring matter, because the manufacturers of these panels search for sell also the control board/software to use the panels... so, it is possible this is the main issue........ but.... we can reach the goal: make your panels work properly. Patience, and WRITE these 17-19 commands, one by one...

Nolicx, 2024-08-13:
All right, i will take note and pictures of all the outputs later today or tomorrow.
Thank you again for your help. :)

Nolicx, 2024-08-14:
Hey,
sorry for the late reply, i was busy.

Here are the results of trying out all multiplexing modes using:
```
for MULTIPLEXING in {0..20}
do
    echo "Running demo with Multiplexing=$MULTIPLEXING"
    sudo ./demo -D0 --led-rows=32 --led-cols=64 --led-chain=1 \
    --led-slowdown-gpio=5 --led-multiplexing=$MULTIPLEXING
done
```
#### 0
[image]
- No Errors

#### 1
[image]
- No Errors

#### 2
[image]
- No Errors

#### 3
[image]
- No Errors

#### 4
[image]
- No Errors

#### 5
[image]
- No Errors

#### 6
[image]
- No Errors

#### 7
[image]
- No Errors

#### 8
[image]
- No Errors

#### 9
[image]
Error Example: `Error in PixelMapper: (32, 0) -> (-1, 3) [range: 256x8]`

#### 10
[image]
Error Example: `Error in PixelMapper: (0, 31) -> (4, 16) [range: 128x16]`

#### 11
[image]
- No Errors

#### 12
[image]
- No Errors

#### 13
[image]
- No Errors

#### 14
[image]
- No Errors

#### 15
[image]
- No Errors

#### 16
[image]
Error Example: `Error in PixelMapper: (40, 0) -> (-1, 9) [range: 128x16]`

#### 17
[image]
- No Errors

#### 18
[image]
- No Errors

#### 19
[image]
Error Example: `Error in PixelMapper: (0, 16) -> (-8, 0) [range: 256x16]`

#### 20
[image]
- No Errors

So to me the most promising multiplexer mode seems to be 0, however only half of the panel is used and the rotating cube seems to be shifted up or down, which is why i think, that i need some other kind of multiplexing.

Here is some additional verification, that the whole panel is useable:
When i enter:
```
sudo ./demo -D0 --led-rows=64 --led-cols=64 --led-chain=1 --led-slowdown-gpio=5 --led-multiplexing=0
```
The whole panel is used:
[image]

Thanks in advance for your help.

closed 2025-11-27 by marcmerlin