hzeller/rpi-rgb-led-matrix #1644: 64x64 Matrix Only Displays Red on Demo

RenAaron, 2024-04-12:
I am experiencing an issue with my [64x64 RGB LED Matrix](https://www.mouser.com/ProductDetail/Adafruit/4732?qs=sGAEpiMZZMu%252BmKbOcEVhFX4GTH4cr2Z%252BGu7cNJkzYHQImobYpS8Dyw%3D%3D) where the demo files only produce red colors (or green, but never anything in between) 

I am driving it with a Raspberry Pi Zero WH and using the[ Adafruit RGB Matrix Bonnet](https://www.adafruit.com/product/3211) with the appropriate short between the "middle pad" and the "8 pad" on the back of the bonnet. 

I am running the command: ./demo -D 0 --led-rows=64 --led-cols=64 --led-gpio-mapping=adafruit-hat --led-slowdown-gpio=1

Any advice/pointers appreciated o7 

https://github.com/hzeller/rpi-rgb-led-matrix/assets/73759943/c711223c-e542-4985-8741-71281589d00b

davemaster, 2024-04-17:
Greetings,

Looks like your panel has only RED color... 
FIRST... check your connections again... SECOND the power supply, use one with at least 5A power... 

THE BEST (for me) it's test those new panels, I don't use any HUB75 board, just plug using common cabling at regular way

check this out [wiring](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md)

RenAaron, 2024-04-19:
> Greetings,
> 
> 
> 
> Looks like your panel has only RED color... 
> 
> FIRST... check your connections again... SECOND the power supply, use one with at least 5A power... 
> 
> 
> 
> THE BEST (for me) it's test those new panels, I don't use any HUB75 board, just plug using common cabling at regular way
> 
> 
> 
> check this out [wiring](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md)
> 
> 
> 
> 
> 
> PD - Share a picture of your HUB75 board or connections to your panel.

It was the power supply! Thank you! This solved my issue!

closed 2024-04-19 by RenAaron

davemaster, 2024-04-19:
> It was the power supply! Thank you! This solved my issue!

Good for You. Don't forget to share a picture/video of your panel working properly. Thanks in advance.

RenAaron, 2024-04-19:
:) 

https://github.com/hzeller/rpi-rgb-led-matrix/assets/73759943/7c0a2379-748f-4aea-8b22-7347e17b4b6a