hzeller/rpi-rgb-led-matrix #1646: No output on RPi 3B+ despite working perfectly on RPi Zero 2

jakubgawlowski, 2024-04-18:
Hi! 

I decided to upgrade my LED display and swapped RPi Zero 2 for 3B+. The problem is that there's no output on the latter, the display is dark and I can't figure out why. 

My display is made of four P4 32x64 HUB75 matrices in one chain. I'm using the same SD card with Raspbian Lite on both Raspberries. I double checked my wiring (wired directly to the Pi, without any hats or adapters). When I run demos with `--led-show-refresh` flag I can even see the refresh rate in the terminal, even though the matrix doesn't show anything.

davemaster, 2024-04-19:
Greetings,

When You upgraded your LED display, use the same "interface" (adafruit, regular, etc.) or just use cables?

If it is so, check your cables, for open cable, DO THIS first, put back your LED display on RPi Zero 2, it still working...? Best regards

PD. A picture of your connections and the command will be great.

jakubgawlowski, 2024-04-19:
Thank you for your reply :)

I'm not using any hats like the Adafruit one, just jumper wires and flat cables wired according to [hzeller's tutorial](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md). I replaced all of the cables used with RPi 3B+ and still nothing. When I put back my RPi Zero 2 the display works just fine.

Since I try to run demo first on the RPi 3B+, the command usually looks like that
`sudo ./demo -D 0 --led-gpio-slowdown=3 --led-cols=64 --led-rows=32 --led-chain=4 --led-no-hardware-pulse --led-brightness=50`

Never any errors, just blank display.

Connections:
[image]
[image]

The only difference is that in RPi 3B+ I skipped the jumper wires for two unused GNDs