hzeller/rpi-rgb-led-matrix #1647: Unknown panel don't work correctly

xorkrus, 2024-04-18:
Hello. Unknown panel, resolution 104x52. Chips dp3216+2013ep. I just can't configure the image output. The best thing came out in the photo - half the lines at the top, half at the bottom and interlaced. Tell me what am I doing wrong?
Command:
`sudo ./demo -D0 --led-row-addr-type=0 --led-scan-mode=1 --led-brightness=10 --led-slowdown-gpio=2 --led-gpio-mapping=regular-pi1  --led-chain=1 --led-parallel=1  --led-multiplexing=0 --led-pwm-dither-bits=0 --led-rows=52 --led-cols=104`
[image]
(it's a moving cube)

davemaster, 2024-04-19:
Greetings,

First, check wiring, twice. Then, change in --led-multiplexing option, from  1 to 14 (I think), ie, **--led-multiplexing=1, then --led-multiplexing=2, etc...**

Also, if You're using a RPi 1, DON'T use --led-slowdown-gpio=2, this is for fastest RPi's. try this command:

sudo ./demo -D0  --led-gpio-mapping=regular-pi1  --led-chain=1 --led-parallel=1  **--led-multiplexing=[1..14]**  --led-rows=52 --led-cols=104

xorkrus, 2024-04-19:
Thank you. Indeed, the main mistake was the lack of contact of the OE wire. This solved part of the problem. 
Now the image is there, but it doesn't look right. If you specify a resolution of 56x112 (52x104 is the physical size), the image becomes more correct. However, the central 8 points (physically these are the last 4 points on the left side and the first 4 points on the right side) work with an incomprehensible shift.
Launch command:
`sudo ./clock --led-gpio-mapping=regular-pi1 --led-chain=1 --led-multiplexing=1 --led-rows=52 --led-cols=104 --led-brightness=7 -f ../fonts/10x20.bdf -d 'OOOOOOOOOOOOO'
`
52x104
[image]
56x112
[image]
[image]
Multiplexing 1 is the best. The remaining options completely break the image.

davemaster, 2024-04-22:
Some of those panels are made putting together 2 or more standard (16x32, 16x16, 32x32, etc) panels, take a picture to the back of your panels, closest to try read the ICs

xorkrus, 2024-04-22:
The microcircuits correspond to the panel markings. They are filled with varnish and it is impossible to photograph the markings. I was able to read the markings of several microcircuits: CHIPONE ICND2013EP, DP3216, 74HC24x. A general photo of the panel is below.
[image]

xorkrus, 2024-04-27:
At least point me in which direction to look for solutions to problems. I'm not good enough at programming to write the code myself, but I can try to google some clever ideas. But I don’t even understand what is needed.

davemaster, 2024-04-27:
I think a new pixel mapping is needed... at this time, this panel is not compatible with this library...

board707, 2024-07-09:
I have already dealt with such panels. The driver used on it has 16 outputs, but the width of the panel is not a multiple of 16. Therefore, the 8 outputs of one of the drivers in the middle of the panel are not connected to anything.
You need to work with the panel as if it were 112x52, but simply fill the 8 dots in the middle of the line with zeros

xorkrus, 2024-07-16:
Thank you. This is useful information. I couldn’t fix the program on my own, now I’m looking for friends to solve this problem and this information will be useful to them.

xorkrus, 2025-01-05:
Nobody wrote fixes for such matrix? I still don't have enough experience :(

board707, 2025-01-06:
I did, but not for the hzeller library

board707, 2025-01-06:
I think the buying another panel is a best option for your case.

xorkrus, 2025-01-06:
I don't want such a panel so much that I'd spend money on it. I just have a few with dead pixels, I'd like to use what I have. It's either RPi1 and RP2040 on python, or esp32c3. But for now, this repository is the most suitable and working one.

board707, 2025-01-06:
> It's either RPi1 and RP2040 on python

My library also works on RP2040 (C++)

xorkrus, 2025-01-21:
Я поимею ещё наглости и попрошу указатель, где в вашем коде функция, которая упоминается в [этом](https://github.com/hzeller/rpi-rgb-led-matrix/issues/1647#issuecomment-2216741732) комментарии?
На RP2040 передумал, там вайфая нет. Попробую вникнуть и написать/поправить этот реп.

board707, 2025-01-21:
The code is located in DMD_Panel_Templates.h file, lines 699-789.
However, this code is unlikely to be useful to you, since it is very specific to my library.
You should be guided by the general principle - you should pass 8 more points to each line of the panel than the length of the row - that is, 112 instead of 104. Of these, eight points in the middle of the row will not be displayed and may contain random data.

xorkrus, 2025-01-21:
I think I'm starting to understand the task. I need to form a line in the mapper "{specified physical length}+8" and "fill 53 to 60 with zeros". and send the resulting array to the panel.
I tried the opposite - I specified the panel size of 112 physical points and filled the middle 8 with zeros, but it turned out even worse than before.