hzeller/rpi-rgb-led-matrix #1648: P3 64x64 matrix (P3-32S-HL1.2) missing red on second half of panel

zango-me, 2024-04-21:
Hi,

i got a few P3 panels (part no 2PH2962EA1 // [https://de.aliexpress.com/item/1005005727115453.html](https://de.aliexpress.com/item/1005005727115453.html)) and **they do not display red on the second half of the panels** (the side away from the connectors)
Everything else looks okay tho, and the behaviour is across all 6 panels I got.

I checked all available options for the following parameters (not every combination tho, caus that would take ages):
```
--led-gpio-mapping
--led-multiplexing
--led-pwm-bits
--led-scan-mode
--led-row-addr-type
--led-pwm-lsb-nanoseconds
--led-pwm-dither-bits
--led-panel-type
```
and both configurations for the E address pin as well but nothing made it better (most changes only broke it more).

This is my current command I'm using to run the demo:
`sudo examples-api-use/demo --led-rows=64 --led-cols=64 --led-parallel=1 --led-chain=1 --led-brightness=30 --led-rgb-sequence=RBG --led-limit-refresh=60 -D4`

The Chip nearest to the connector is a **DP245C**

_Does anyone have an idea what the problem could be?_

[image]

[image]

[image]

zango-me, 2024-04-21:
So I tried it with a **Pi1 model B+** which I had still laying around and it seems to work with that (except some crazy slowdown when I address all 6 panels, which is expected due to the performance of the system I guess), is this some strange behaviour of the Pi4 or is there a GPIO broken on mine?

_If so can I remap the broken GPIO somehow? (and how do I identify it?)_

PS:
I use respbian lite on both, 32bit on the Pi1, 64bit on the Pi4 and I followed the troubleshooting steps (disableing audio driver and uninstalling the packages mentioned there) on both PIs

davemaster, 2024-04-22:
Greetings,

You have only of this panel? Have another one like 32x32?

Also, change this:

sudo examples-api-use/demo --led-rows=64 --led-cols=64 --led-parallel=1 --led-chain=1 --led-brightness=30 --led-rgb-sequence=RBG --led-limit-refresh=60 -D4

to

sudo examples-api-use/demo --led-rows=64 --led-cols=64  -D4

zango-me, 2024-04-22:
Hi,
thanks for your quick reply

---

> You have only ONE unit of this panel? Have another one like 32x32?

I have only 6 of the same type of panel, they all do the same but allso all work with the other Raspberry Pi (as mentioned) for some reason

---

> sudo examples-api-use/demo --led-rows=64 --led-cols=64 -D4

That's what I started with when trying the panels out, but has the exact same behaviour.
(Well, colors are wrong cause my panels are RBG and not RGB but red is not working on the one half with both configurations, so it must be an addressing issue or something similar)

---

any other ideas?

davemaster, 2024-04-22:
ok

add in Rpi4 the led-gpio-slowdown option

now:

sudo examples-api-use/demo --led-rows=64 --led-cols=64 -D4 --led-gpio-slowdown=4 

test, between 2-4 (even 5)

[read this](https://github.com/hzeller/rpi-rgb-led-matrix?tab=readme-ov-file#gpio-speed)

zango-me, 2024-04-22:
That option dosn't help, sadly

davemaster, 2024-05-29:
With the RPi 4,  are You using the SAME EXACT connections than the used with Rpi 1? U are using THE SAME adapter, cabling for both? Show pictures of BOTH

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin