hzeller/rpi-rgb-led-matrix #1649: QiangLi P6 48x24 working not correct

Rpsl, 2024-04-23:
Hello. 

I have a QiangLi P6 Outdoor matrix, 48x24 diodes. I connected it according to the instructions in [wiring.md](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md), but it doesn't work properly.

The label on the matrix says it is ‘sm16208sj + rul5158c’, also on the matrix I see not typical HUB75 pins.

- When I try to run with `./demo -D 4 --led-gpio-mapping=regular`, every odd line lights up. 
- When I try to run with `./pixel-mover --led-gpio-mapping=regular`, I see a couple of pixels with (0,0) and zero pixels with (0,1).

I don't understand if this is a wiring problem or a multiplexer problem. If it is a multiplexer problem, how can I determine the correct positioning?

Can anyone help me with this?

[image]
[image]
[image]

[image]

[image]

davemaster, 2024-04-23:
Greetings,

You forgotten to use in your command these:

--led-rows=<rows>        : Panel rows. Typically 8, 16, 32 or 64. (Default: 32).
--led-cols=<cols>        : Panel columns. Typically 32 or 64. (Default: 32).
--led-multiplexing         : panel scan, test from 1 to 14 (i think)

[view this](https://github.com/hzeller/rpi-rgb-led-matrix?tab=readme-ov-file#types-of-displays)

davemaster, 2024-04-23:
> @davemaster thanks. i used all possible combination of `--led-multiplexing` and `--led-row-addr-type`. also i used `rows` and `cols` options.
> 
> i not wrote that in issue but i test it.

what exactly you wrote?

Rpsl, 2024-04-23:
> what exactly you wrote?

```
./demo -D 4 --led-gpio-mapping=regular --led-cols=48 --led-rows=24 
   --led-multiplexing=0..18 # i try all options
   --led-row-addr-type=0..4 # i try all options
```

Rpsl, 2024-04-24:
> And, a PICTURE at least of the value of multiplexing that better works ?

Values of `--led-multiplexing` don't give any changes. At some multiplexing values every second line is filled in the matrix, at others it does not lighting matrix.

this is how it looks: 
[image]

ps.
- It's work only with `--led-row-addr-type=0`, with other options matrix not lighting.
- I check it with matrix on another chip and my setup work correct.
- I check it with several other matrix (QiangLi P6 `sm16208sj + rul5158c`) and that problem repeated.

Rpsl, 2024-05-06:
just bought new panels with another chipset and they work perfect

closed 2024-05-06 by Rpsl