hzeller/rpi-rgb-led-matrix #1651: Facing issues  with rgb led matrix 64*32 (Chinnese) with Mega 2560

kshitij283, 2024-04-26:
I was facing issue with rgb led matrix . I have tried different libraries but not able to resolve the issue. In the given image i was trying to print T1 but the out was different. I also do not have any documentation of the led pannel. I have shared the image of Hub75 port also . 
[image]
[image]

kshitij283, 2024-04-26:
It was P5 and was from Qiang Li manufacturer

davemaster, 2024-04-26:
Mega 2560? Are You sure you're talking about this, the[ rpi-rgb-led-matrix github](https://github.com/hzeller/rpi-rgb-led-matrix)? and not of the other from [Adafruit](https://github.com/pkourany/RGBmatrixPanel_IDE)? 

This works with Raspberry Pi SBC 1, 2, 3 & 4, 5 if I can get some free time

kshitij283, 2024-04-26:
I do not have raspberry Pi, I have esp32 and mega 2560

davemaster, 2024-04-27:
> I do not have raspberry Pi, I have esp32 and mega 2560
> […](#)

Here is the wrong place, go to adafruit place, follow FIRST, the wiring, then, Arduino IDE library installation then, an example