hzeller/rpi-rgb-led-matrix #1652: A better way to load a large amount of GIFs in Python (100+)

ThisIs0xBC, 2024-05-09:
Hi, I've created a GIF slideshow application in Python that takes GIFs uploaded to a website (web server is made with Flask and runs completely separately to the led matrix code), and will play them 5 times, then cycle on to the next GIF. The user can upload their own GIFs to the server and select which ones they want in the display queue.

This is the code I used to pre-process the GIFs:
```python
for i in range(5): #Plays GIF 5 times
    for frameNumber in range(len(gifCanvases[0])): #gifCanvases[0] is a list of FrameCanvases returned by "canvas = self.matrix.CreateFrameCanvas()" which make up each frame of the GIF
        self.matrix.SwapOnVSync(gifCanvases[0][frameNumber], framerate_fraction=framerateFraction)
```

Anyways, the way I render them is basically having a dictionary of lists, each dictionary key is a GIF filename, and each key has a value of a list containing:
A list of FrameCanvas objects containing each frame of the GIF
A number representing the GIF "duration"

The first time a GIF is rendered, I preprocess each frame into a 64x64 BMP file, with this code:
```python
gif = Image.open(f"{self.webappRootDir}/basic-flask-app/static/gifs/{gifName}")

for frame_index in range(0, gif.n_frames):
    gif.seek(frame_index)
    # must copy the frame out of the gif, since thumbnail() modifies the image in-place
    frame = gif.copy()
    frame.thumbnail((self.matrix.width, self.matrix.height), Image.Resampling.LANCZOS)
    canvas = self.matrix.CreateFrameCanvas()

    frameImageData = frame.convert("RGB")
    frameImageData.save(f"{self.matrixRootDir}/saved-gifs/{gifName}/frame_{frame_index}.bmp")

    canvas.SetImage(frameImageData)
    canvases.append(canvas)

    #Store new GIF information in canvases dictionary
    self.gifFileCanvases[gifName] = [canvases, gif.info['duration']]
    
    #Close the gif file to save memory now that we have copied out all of the frames
    gif.close()
    print("Processed GIF from original file and saved data to file")
```

Once the GIF has been processed into its individual 64x64 frames, the next time I run the code or need to load that GIF it just loads directly from the BMP files, so I can skip the scaling down to 64x64 (which saves a BUNCH of loading time, its basically instant loading this way)

However, I still need to call:
```python
canvas = self.matrix.CreateFrameCanvas()
canvas.SetImage(Image.open(f"{self.matrixRootDir}/saved-gifs/{gifName}/frame_{frameFileNumber}.bmp"))
canvases.append(canvas)
```
for EVERY frame of EVERY GIF! Here is the full code of just loading a GIF from the 64x64 BMP files:

```python
gif = Image.open(f"{self.webappRootDir}/basic-flask-app/static/gifs/{gifName}")

directory = os.fsencode(f"{self.matrixRootDir}/saved-gifs/{gifName}")
numberOfFrames = len([f for f in os.listdir(directory)if os.path.isfile(os.path.join(directory, f))])
        
for frameFileNumber in range(numberOfFrames):
    canvas = self.matrix.CreateFrameCanvas()
    canvas.SetImage(Image.open(f"{self.matrixRootDir}/saved-gifs/{gifName}/frame_{frameFileNumber}.bmp"))
    canvases.append(canvas)

    self.gifFileCanvases[gifName] = [canvases, gif.info['duration']]
    
gif.close()

print(f"Loaded GIF Data from File: {gifName}")
```

The above code is essentially a modified version of https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/bindings/python/samples/gif-viewer.py

This creates an issue, as the canvas objects generated when calling CreateFrameCanvas() don't appear to be deleted until the program is killed. After 500 and 1000 calls to it, it shows this warning:
```
CreateFrameCanvas() called 500 times; Usually you only want to call it once (or at most a few times) for double-buffering. These frames will not be freed until the end of the program.

Typical reasons:
 * Accidentally called CreateFrameCanvas() inside your inner loop (move outside the loop. Create offscreen-canvas once, then re-use. See SwapOnVSync() examples).
 * Used to pre-compute many frames (use led_matrix::StreamWriter instead for such use-case. See e.g. led-image-viewer)
```
And it then shows this again for every consecutive 500th call to the function.

The led_matrix::StreamWriter functionality is not exposed/doesn't have a Python binding, so is there a more optimal way to achieve what I'm trying to do without thousands of calls to CreateFrameCanvas?

I tried serialising the canvas objects via pickle, but it tells me the object is not able to be serialised (I assume because its a C type object with non standard fields?), any suggestions?

Thanks in advance.

ThisIs0xBC, 2024-05-09:
Just had an idea, would it be possible to just create a sinuglar frame canvas instead, and just utilize PILs "SetImage" functionality like this:

```python
gif = Image.open(f"{self.webappRootDir}/basic-flask-app/static/gifs/{gifName}")

directory = os.fsencode(f"{self.matrixRootDir}/saved-gifs/{gifName}")
numberOfFrames = len([f for f in os.listdir(directory)if os.path.isfile(os.path.join(directory, f))])
    
canvas = self.matrix.CreateFrameCanvas()
   
for frameFileNumber in range(numberOfFrames):
    
    canvas.SetImage(Image.open(f"{self.matrixRootDir}/saved-gifs/{gifName}/frame_{frameFileNumber}.bmp"))
    canvases.append(canvas)

    self.gifFileCanvases[gifName] = [canvases, gif.info['duration']]
    
gif.close()

print(f"Loaded GIF Data from File: {gifName}")
```

Then when rendering the GIFS I can just iterate each frame and run:
```python
for frame in canvases:
    matrix.SetImage(frame)
```
Where "frame" is a "FrameCanvas" object we created when processing the GIF before?

ThisIs0xBC, 2024-05-10:
Ok, update. I tried the above, and it partially works.

I now just save the bitmap files as usual (like shown above), and then when rendering each GIF, I first load all the BMP with Image.open(), passing the each BMP file (each frame of the GIF) as an argument, then saving all the results of Image.open() into a list. So I end up with a list of Image objects.

Then, when rendering, I iterate that list of Image objects and run:
```python
self.canvas.SetImage(listOfImages[frameNumber])
self.matrix.SwapOnVSync(self.canvas, framerate_fraction=framerateFraction)
```

This works, but produces noticeable flickering when the GIF restarts (as I play them 5 times each).

Any ideas? I just have the rendering code wrapped in a `for i in range (5)` loop for playing them 5 times.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828