hzeller/rpi-rgb-led-matrix #1653: Noob needs help with their setup

oryz0, 2024-05-10:
Hi community!
I'm a passionate noob trying to learn electronics, and I've spent many hours trying to solve my issues on my own. Now I'm reaching out to you:
I've purchased 3 [LED panels 64x32](https://www.amazon.fr/dp/B0C4LDQK2S/ref=pe_27091421_487030221_TE_SCE_3p_dp_i1), [Raspberry Pi 4B](https://www.amazon.fr/dp/B07TGBLK33/ref=pe_27091421_487030221_TE_SCE_3p_dp_i1), [a 5V 15A power supply](https://www.amazon.fr/dp/B0B49ZN1LF/ref=pe_27091421_487030221_TE_SCE_dp_i2), and 1 [Adafruit HAT](https://www.amazon.fr/dp/B00SK69C6E/ref=pe_27091421_487030221_TE_SCE_3p_dp_i1).
I've carefully followed the guide provided on the website, but I can't seem to display anything other than glitches and color bands like in the photo :
[image]
However, my Raspberry Pi seems to be sending information to the LED panels properly.
I bought a [LED-matrix controller](https://www.amazon.fr/dp/B07YL1J9CW/ref=pe_27091421_487052621_TE_item_image) thinking that the problem might be with the Adafruit HAT.
What could I have done wrong? Bad wiring? Incorrect commands?
Thank you in advance for your invaluable help!

davemaster, 2024-05-13:
Before I help You, I think you must spend a couple more of hours reading the guide, right?

Then, keep in mind this; read, I mean, READ carefully the Adafruit guide of your HAT, [this link](https://learn.adafruit.com/adafruit-rgb-matrix-plus-real-time-clock-hat-for-raspberry-pi?view=all) in a single page.

In some cases, the rpi-rgb-led-matrix is compiled for "regular" wiring, but you need to set "adafruit-hat". Also, You need to try (if You don't know already) --led-multiplexing=0..to.. 19 (the type of your RGB Panel), and slow the GPIO in your RPi 4, --led-slowdown-gpio=<0..4>, then use in demo as follow:

sudo ./demo -D 0 --led-gpio-mapping=adafruit-hat --led-rows=32 --led-cols=64 --led-chain=3 --led-no-hardware-pulse --led-multiplexing=1 

It doesn't work yet? Another hour of reading, soon you get the right parameters, keep trying --led-multiplexing with other values until 17

Best Regards

oryz0, 2024-05-29:
Hello !

Thanks a lot for your help. I read the guide several times. I didn't weld the HAT, so I focused on making the connections.
Once I assembled all the components as described in the guide, I unfortunately got stuck at the same point.
I applied your code line with all possible led-multiplexing and slowdown GPIO combinations.
The information is sent correctly, but nothing is visible.
In my previous photo, I think it was just an electrical issue. I saw this line error before unplugging the GPIO pin.

In the guide, I saw the instruction to weld jumper “E” to pad 8, but there is very little information about values 8 and 16. I don't know what information to look for in my panel settings since they don’t come from the Adafruit shop. To be safe, I welded jumper 8 as described in the guide.

Another detail is that there was a short circuit in my first assembly at the right GPIO ports. This is why I bought a new LED Matrix controller, thinking I had damaged my HAT. However, the commands don’t work with this controller either.

The guide is very clear, and I followed it carefully. Now I don’t know what to do to fix my setup.

[image]

oryz0, 2024-05-30:
Hello, ok I have a RPI 4B and 5v 15a alimentation
Where can I find REGULAR[1] in the wiring guide ? I don't see on this [page](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md)
I make video : 

https://github.com/hzeller/rpi-rgb-led-matrix/assets/169474469/230c2ccd-0a90-4189-912d-5baa7b9d4a94

Thanks a lot for your help

davemaster, 2024-05-31:
> Hello, ok I have a RPI 4B and 5v 15a alimentation Where can I find REGULAR[1] in the wiring guide ? I don't see on this [page](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md) I make video :
> Led_Pannel_Issue_1.mp4
> 
> Thanks a lot for your help

BUT YOU FORGOT to share the output in RGB Panel of each command....

I note you feed the 5V to panel FROM Adafruit HAT; no, feed power from the power supply directly, I AM NOT SURE if the panel supplies enough current to panel... 

Waiting your RGB PANEL OUTPUT in video... I will be all night long working...

oryz0, 2024-05-31:
I think it is not important to share the output because nothing appears on the LED panel. I have tried to feed power directly to the power supply, but there is nothing there too. I know it is annoying to help an inexperienced person. Thank you for taking the time to help me.

davemaster, 2024-05-31:
> I think it is not important to share the output because nothing appears on the LED panel. I have tried to feed power directly to the power supply, but there is nothing there too. I know it is annoying to help an inexperienced person. Thank you for taking the time to help me.

well, if you went so far filming that video, a little seconds more.. BUT, if NOTHING is showed in the panel.... YOU MUST FIRST, check your:

1. POWER SUPPLY, TRY WITH ANOTHER... for 64x32, 5V 5A at minimum
2. Check YOUR CONNECTION, use the REGULAR in wiring, for my 64x32 panel, I have NO NEED to do some hack/mod to my Electrodragon RPi adapter, (but yes I need[ for another 64x64 32S panel](https://youtu.be/sBRi85QoGuY))

oryz0, 2024-06-06:
Hi,
Unfortunately, I don't have another backup power supply. However, if that's the question, there is indeed electricity flowing through the panels. My first photo in this conversation shows an electrical bug; the image isn't good, but there is a connection when I touch it with my hands :
https://github.com/hzeller/rpi-rgb-led-matrix/assets/169474469/5f9e05f7-da57-4e5a-b900-0a468db563dd

I replaced the adafruit-hat with regular, and I also formatted my SD card and reinstalled the library again, but nothing happens.

davemaster, 2024-06-06:
Power supply, 5A at min for each panel

oryz0, 2024-06-07:
Presently I have 5V 15A power supply for one pannel, that’s enough, isn’t it ?

davemaster, 2024-06-08:
> Presently I have 5V 15A power supply for one pannel, that’s enough, isn’t it ?

NOW, try; if it doesn't work yet, keep trying:
1. github this to another new folder
2. check your connection, don't use any hat, use "REGULAR" cabling one
3. test that new 15A 5V power supply
4. if it's possible, use a rgb led panel and RPi, both tested (remember, those GPIO also can be broken)

God Bless You

oryz0, 2024-06-26:
Hi,

What do you mean by "new folder"? A new issue in rpi-rgb-led-matrix forum ?

I am unable to understand regular wiring, which is why I bought a hat and then a Matrix controller to avoid dealing with it. I do very little electronics; what I'm passionate about is graphic animations on LED panels.
I tested the new power supply, nothing changes.
How can I test or diagnose my panels and my RPI?

Best regards, and thank you again for your patience

davemaster, 2024-06-26:
then, purchase an [ElectroDragon active board](https://youtu.be/sBRi85QoGuY)... a very good option and be free from wiring again..

closed 2025-11-27 by marcmerlin