hzeller/rpi-rgb-led-matrix #1654: FYI

MiguelElBigodon, 2024-05-16:
I have the following error:
FYI: not running as root which means we can't properly control timing unless this is a real-time kernel. Expect color degradation. Consider running as root with sudo.

How can I solve this problem?

I have the adafruit matrix hat+real time clock board, two 16x32 matrices and a Raspberry Pi 4b.

hzeller, 2024-05-16:
> I have the following error:
> FYI: not running as root which means we can't properly control timing
> unless this is a real-time kernel. Expect color degradation. Consider
> running as root with sudo.
>
> How can I solve this problem?

Run with sudo

I have the adafruit matrix hat+real time clock board, two 16x32 matrices

MiguelElBigodon, 2024-05-16:
The problem:

sudo python3 /home/migasboss/rpi-rgb-led-matrix/bindings/python/samples/runtext.py Press CTRL-C to stop sample Traceback (most recent call last): File "/home/migasboss/rpi-rgb-led-matrix/bindings/python/samples/runtext.py", line 33, in <module> File "/home/migasboss/rpi-rgb-led-matrix/bindings/python/samples/samplebase.py", line 76, in process File "/home/migasboss/rpi-rgb-led-matrix/bindings/python/samples/runtext.py", line 14, in run File "graphics.pyx", line 32, in rgbmatrix.graphics.Font.LoadFont Exception: Couldn't load font ../../../fonts/4x6.bdf

Fonts imagem
[image]

Runtext:
from samplebase import SampleBase 
from rgbmatrix import graphics
 import time class
 RunText(SampleBase): def __init__(self, *args, **kwargs): super(RunText, self).__init__(*args, **kwargs) self.parser.add_argument("-t", "--text", help="The text to scroll on the RGB LED panel", default="MECATRONICA") def run(self): offscreen_canvas = self.matrix.CreateFrameCanvas() font = graphics.Font() font.LoadFont("../../../fonts/4x6.bdf") textColor = graphics.Color(255, 255, 0) pos = offscreen_canvas.width my_text = self.args.text while True: offscreen_canvas.Clear() len = graphics.DrawText(offscreen_canvas, font, pos, 25, textColor, my_text) pos -= 1 if (pos + len < 0): pos = offscreen_canvas.width time.sleep(0.05) offscreen_canvas = self.matrix.SwapOnVSync(offscreen_canvas) # Main function if __name__ == "__main__": run_text = RunText() if (not run_text.process()): run_text.print_help()

MiguelElBigodon, 2024-05-21:
I'm having the problem mentioned above, which says I don't have the bdf files but they're all in the folder where they should be.

davemaster, 2024-05-21:
A little more reading of "README.MD" in this repository, and problem solved... a little more. An advice, before using python scripts, try using the demo within the examples-api-use folder.

Best regards

PD. IN python, the code must be PERFECTLY indented, otherwise... ERROR everywhere...

closed 2025-11-27 by marcmerlin