hzeller/rpi-rgb-led-matrix #1655: Weird Noise and Flickering

IAteYourCookies, 2024-05-19:
As in the title i am experiencing some weird noise that gets worse over time when i start the program.

How it looks: 
https://youtu.be/2eZnr0oL_mg

You can see the Text creeping in from the right and then get overtaken by the noise.

The setup tutorial i used.
https://youtu.be/xSwoM3M24W8?si=GniFOo2iN2pi77WL

I also used this back in the day (like half a year ago) and it worked. But since, ive tried 5 times to get it to work, without success (5 complete reinstalls of Raspbian in 64 and 32 bit.).
The video i recorded is on a 32 bit legacy raspbian OS.

davemaster, 2024-05-19:
The SAME using 

$ sudo ./demo ......................  ???

IAteYourCookies, 2024-05-20:
Here is me trying the demo: (also didn't work)
[image]
[image]

I am using a raspberry pi 4 with 4Gb ram.
And I am also using the same panel.
As said, I got it working half a year ago with the images displaying perfectly fine. But now I can't seem to get it to work again.

davemaster, 2024-05-21:
Friend, YOU 'RE NOT using --led-multiplexing; NOT ALL the panels are manufactured by one company, there are a lot of them, use many values : --led-multiplexing=1, then  --led-multiplexing=2,  --led-multiplexing=3 , etc... I think know there are 18 or 19 compatible panels.... try again... and READ the README.md in the rpi-rgb-led-matrix github... it' s all there

IAteYourCookies, 2024-05-21:
Sadly I am using the same panel. This is my result without any multiplexing and you can see the panel clearly trying to show the text:
https://youtu.be/l-kISpJOLiQ?si=cGpfd_99H7YlC4Bj

Whenever I tried all the multiplexing options (I tried all 18) there would always be a pixel gap in the middle of the matrix like this:
[image]

And as I have already said twice: It used to run perfectly fine (with the exact same command)

davemaster, 2024-05-21:
For this kind of problems, the BASIC solutions. Do this, if I am right, your interface is the adafruit- hat, right? STOP using that hat, and wire the rgb panel using this guide:  [wiring guide](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#connections-per-chain); doing that, we will discard as defected that adafruit-hat (could it happen).

Try that, and use --led-gpio-mapping="regular"; record a video an share the results 

Best of luck

IAteYourCookies, 2024-05-21:
I have tried to follow the wiring guide, but it didn't really work.
(Showed absolutely nothing on the display)
I will try to do it again tomorrow.

davemaster, 2024-05-21:
Try, remember, with --led-gpio-mapping = "regular", YOUR POWER SUPPLY, over 5 A (amperes) even a more powerful to be sure, also, the Rpi GPIo ports can be broke... the only way to test that, is with a project that turn on/off a led plugged into it, there are a lot online

IAteYourCookies, 2024-05-21:
I've now tried 4 times to get it to work with the manual cables.
I even flipped them to make sure I didn't do it wrong.
Only thing I saw were some very dim pixel lines when I flipped the wiring.

Else there is just a blank matrix.

[image]

IAteYourCookies, 2024-05-21:
[image]

davemaster, 2024-05-21:
Use another power supply, at least of 5A power

I use one of 10A, or also use an ATX PC Power supply for prototyping

hzeller, 2024-05-22:
The artifacts look a bit like that GND is not connected between the Raspberry Pi and the Panel. Also it looks like the Panel is not properly powered at all, as the colors tend to be in the red range which indicates low voltage. Also the 'dim pixels' indicates that. This typically happens if the panel is only receiving parasitic power through the digital outputs from the Pi, but not from the power inlet on the back of the panel.

The supply wired to the power inlet of the panel (typically some sort of terminal in the center of the panel on the back) should be in the 5A or more range (5A ok for a single panel). Use proper cabeling (at least 0.75mm²  per panel from your 5V power supply). The pictures indicate that there is no good power.

The power supply for the Raspberry Pi can be different (and typically is, as you power the pi via the USB). This makes it important that the Ground (minus) of both are connected properly, as the panel interprets the digital signal relative to that level. If you followed the [wiring diagram](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md), the GND connections are marked there.

Also, if you directly connect the Pi to the panel without signal level converters is also a likely that the panel has trouble interpreting the 3.3V digital signal on the 5V digital signal input. Reduce the data transmission speed by increasing the `--led-slowdown-gpio` value.

IAteYourCookies, 2024-05-22:
Can I assume that this wiring diagram for HUB75 is correct?
[image]
I want to use this wiring Diagramm since the one in the readme doesn't show the Grounds and the E and my matrix also doesn't show them.

If yes then I would only have to hook up all the corresponding smiling emojis of the Pi to the correct pins on my matrix
[image]

I ordered some more wiring cables to directly hook up the pins without having to use 2 cables for one connection (then i can also skip the ribbon cable) and I will try it out again when I receive my order.

davemaster, 2024-05-25:
NO, You don't understood, 

[image]

the [1] in this repository wiring means the 1st chain of panels, [2] the 2nd,  [3] the 3rd.... 

SO, only keep in mind, the order of the signals in a HUB75, then PAIR for 1 panel, all hose with  [1] R1, [1] G1 .... etc

IAteYourCookies, 2024-05-25:
I was gonna do it like this anyways. The photo just happened to have some more info than needed.

IAteYourCookies, 2024-05-25:
I wired it up and since it still doesn't show anything
[image]

I suppose I need to get a power supply with 5V and 5A

hzeller, 2024-05-25:
how do you power the panel now ? (the power connector typically in the center of the panel)

IAteYourCookies, 2024-05-25:
I got it to work! I manually wired everything up with just direct cables.

https://youtu.be/VJfSZU02kpg?si=0z3fkVQY3G3oLEXl

I am still using the 5V 3.3A PSU
as shown in my photo
(I had to twist it around in my power socket for some reason. The USB-C and the Plug have to be in this specific orientation for it to work. Idk why ... but it works now)
[image]

I'm using a USB-C to Barrel cable btw.(1 end is USB and other one is Barrel)

And I'm hooking it up to the power cable like this:
[image]

hzeller, 2024-05-25:
For a single panel, 3.3A is probably sufficient. Bummer that the cable is unreliable ...

davemaster, 2024-05-25:
Friend, I told You try with ANOTHER power supply, 5A or more... nominally the case of AC Adapter can say ANYTHING, but the REAL POWER could be LESS.. specially since all is manufactured in China...