hzeller/rpi-rgb-led-matrix #1656: Matrix Displaying 2 rows after every 2 rows

JAM-200, 2024-05-23:
[image]
 ive tried with another pannel and they both do this so the panel is working, I've also tried all the demos and a python print text and they all do this same thing. i am using a raspberry pi zero

Led Board
https://www.amazon.ca/gp/product/B0B5N5HPKX/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1

davemaster, 2024-05-23:
Rows issue. PLEASE, always share your command, and picture of your wiring or hat.

Best regards

PD. try many values of --led-multiplexing= 1... 18 to find you panel type

getgotbois, 2024-05-23:
Hi sorry for not showing the command or the wiring, I tried all the different multiplexing options and the default looks the best still.
[image]
[image]

getgotbois, 2024-05-23:
Oh and the panel power is connected to a wall power extender usb, and the pi is connected to a battery bank (my usbs under my desk shorted out from a speaker project not too long ago)

davemaster, 2024-05-23:
Sadly, in each multiplexing try You doesn't post the RGB Panel output... it can tell a lot; at least that with better results
try to use in your command:

$ sudo ./demo -D 0 --led-rows=32 --led-cols=32 --led-multiplexing=**_1-1x_** --led-gpio-mapping=adafruit-hat-pwm

some panels manufacturer use x2 32x32 panels together... 

Also, put some pictures of those panels works fine

JAM-200, 2024-05-23:
after playing around with jumper pins to connect it I'm just gonna guess its a hardware issue with the hat, thanks for your help though

closed 2024-05-23 by JAM-200

davemaster, 2024-05-24:
Here my solution, just in this days I was trying to probe my own 64x64 p3 screen, and it doesn't worked with my adapter (electrodragon). But thanks GOD, it was possible with a little mod, build-in with the adapter:

details [here in my video](https://www.youtube.com/watch?v=sBRi85QoGuY)