hzeller/rpi-rgb-led-matrix #1658: video-viewer and led-image-viewer utils displaying noise

jillrene, 2024-05-27:
Hi! I have to 32x32 panels connected to a raspberry pi 3 via a HAT and have been able to install everything as expected. The demo code and python examples all run very well for me. I compiled video-viewer and led-image-viewer and both only display green horizontal flickering lines when being run. Unsure as to what would cause this as I'm running the examples from the README. 
[image]

davemaster, 2024-05-28:
AND, your command ( $ sudo ------------etc------------etc----------ectc ) ?????? Thanks in advance

jillrene, 2024-05-28:
im running sudo ./video-viewer --led-chain=2 --led-parallel=1 --led-limit-refresh=200 -T3 file.mp4
The output looks correct and not throwing any errors but not actually displaying image

davemaster, 2024-05-28:
I suppose, those panels are Adafruit 32x32 1/16 Scan? Why you are using --led-limit-refresh=200, and WHAT POWER SUPPLY are You using... for x2 (TWO) PANELS, You need at lease, a 5V Power Supply of 10A...

davemaster, 2024-05-28:
try, with ONLY x1 ONE panel...

hzeller, 2024-05-28:
If you're using the Adafruit HAT, you need to pass the `--led-gpio-mapping=adafruit-hat` flag.

jillrene, 2024-05-28:
That worked perfectly thank you so much!

closed 2024-05-28 by jillrene

davemaster, 2024-05-29:
> That worked perfectly thank you so much!

ONE MOMENT, please, Ho do You get your project run? Your solution, PLEASE, share it... Thanks in advance 
(share a video or picture, or both)