hzeller/rpi-rgb-led-matrix #1662: led-image-viewer

Camarg03, 2024-06-04:
Thank you very much in advance for this extraordinary library, I have managed to image a 64X64 matrix and it looks spectacular but I can't get the "video-viewer" to work, I think I have correctly converted the video to webm and it seems that everything is correct but I don't get the image with this code

sudo /home/pi/rpi-rgb-led-matrix/utils/video-viewer -f --led-rows=32 --led-cols=64 --led-parallel=2 --led-chain=1 --led-multiplexing=1 /home/pi/images/myvideo.webm -F -f 

when I exit I get this message:

^CGot interrupt. Exiting
Total of 310 frames decoded

Translated with www.DeepL.com/Translator (free version)

davemaster, 2024-06-04:
Greetings,

Can You share that video or it is personal? Thanks in advance (had You tried with another video?)

Camarg03, 2024-06-04:
[myvideo.webm](https://github.com/hzeller/rpi-rgb-led-matrix/assets/65510303/9c399e8d-fbb4-409c-bcc6-1f304ee411b5)

Camarg03, 2024-06-04:
this is te file

davemaster, 2024-06-04:
Greetings,

I note, you're not using the --led-slowdown-gpio option, use it if you have a Raspberry Pi 2,3,4. Then, bellow a link with a video, I'm using in there a 64x32 RGB Panel, with an Electrodragon active interface (regular) and a Raspberry Pi 3 B+ with Raspbian Buster. I just compiled within "utils" directory the image and video-viewer, and test video-viewer is running just fine.

https://youtu.be/AlCGU3EJ3uM

Add the --led-slowdown-gpio =4 ( or 3 ) missing in your command and try. Share the output of YOUR panel

Camarg03, 2024-06-05:
[image]

Camarg03, 2024-06-05:
the same thing, it doesn't give me an image and when I exit, the frames I generated appear.

i use the same code of you video

Camarg03, 2024-06-05:
thank you very much for replying, tomorrow I will try on another rspberry pi, I hope I have luck, cheers.

Camarg03, 2024-06-05:
i made a test right now in an other raspberrypi 4b, whit the same resuts, the thing is,  i can sed static image perfectly, also gifs, but i dont have lucky whit video.

davemaster, 2024-06-05:
Ok, TRY THIS also, before you try with the sudo ./demo .... command

try to build from ZERO the all library:
1. make clean
2. make all

all this (1 and 2) from the folder rpi-rgb-led-matrix/utils/; so, you re-build (if it is not already builded) video-viewer and image-viewer. Check, the amount of memory for GRAPHICS... from raspi-config