hzeller/rpi-rgb-led-matrix #1666: Noise in second and third panel in chain

francescofact, 2024-06-17:
Hello, 
I've start playing with my rpi and three led panel.
I have setup the Adafruit Hat as in the README (bridged (8 and E) and (4 and 18)).
I still see some flickering noise in pixel that shouldnt be on on the panels that are connected as second and third. 

[Video Youtube](https://youtu.be/yAF_vw3yD2E?si=155PyO9uSuUy-0oP)

Python Code running it:
```
#!/usr/bin/env python
import time
from samplebase import SampleBase
from utils import draw_number

class Identify(SampleBase):

    def __init__(self, *args, **kwargs):
        super(Identify, self).__init__(*args, **kwargs)
        
    def run(self):
        self.matrix.brightness = 30
        canvas = self.matrix.CreateFrameCanvas()

        # border
        for x in range(self.matrix.width):
            for y in range(self.matrix.height):
                if y == 0:
                    canvas.SetPixel(x, y, 255, 0, 0) #R
                elif (y+1 == self.matrix.height):
                    canvas.SetPixel(x, y, 0, 255, 0) #G
                elif x == 0:
                    canvas.SetPixel(x, y, 255, 255, 0) #Y
                elif (x+1)%64 == 0:
                    canvas.SetPixel(x, y, 0, 0, 255) #B
                    canvas.SetPixel(x+1, y, 255, 255, 0) 
        
        # numbers
        for num in range(4):
            base = (30+(64*num),28)
            draw_number(canvas, 3-num, base)

        canvas = self.matrix.SwapOnVSync(canvas)
        while True:
            time.sleep(1000)
            continue

# Main function
if __name__ == "__main__":
    identify = Identify()
    if (not identify.process()):
        identify.print_help()

```

I guess it's a hardware issue as the first matrix connected doesnt have any problem. How can i fix this?
The power is enough for the two display and it's injected in all of them.

davemaster, 2024-06-17:
Greetings,

Had You tried with the panels NOT MOUNTED?, try, one by one

francescofact, 2024-06-17:
@davemaster I tried swapping the order and the first in the chain always work, while second and third always show flickery noise

davemaster, 2024-06-17:
first, check your POWER, each panel, consumes at least 5A, each.. then, those cables, one by one, run a demo with each panel with its cable. You know, python is a more slow (than c, c++) programming language, it's a good thing to try FIRST, using the examples in examples-api-use folder... run them first, and SHARE here, the command, and the picture or video -better- of the result... Thanks in advance

davemaster, 2024-06-17:
also, search in README.md of this repo, about the arguments you can use to avoid this:

[img/text-ghosting.jpg](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/img/text-ghosting.jpg)

francescofact, 2024-06-20:
I manage to fix the problem setting "--led-slowdown-gpio" to 4.
It also improved the refresh by 10Hz.

Thank you!

davemaster, 2024-06-20:
If it's possible, share your command/code (if its changed) and your picture / video output. Thanks

closed 2024-06-21 by francescofact