hzeller/rpi-rgb-led-matrix #1669: 4 RGB pin panels

sub-zerohm, 2024-06-25:
I opened an issue earlier but I closed it because I was wrong about how the scan works 😅.  I have 2 panels but they use a different wiring configuration with 4 RGB pins.  I was wondering if that could be added or if there's a fork that already exists with this.

Here's the pinout, you can see it's an ABC panel with some (not connected) pins

[image]

sub-zerohm, 2024-06-29:
Just reading more on the coding and I think this should be possible.  I'll try and make my own fork and separate the data into four lines instead of two.  I have 2 of these panels equaling 256x256 which should achieve a viewable hertz

nick0016-git, 2024-12-10:
Any success with this? 

I got 48 x 48 6S panel which has the same 4 RGB pins.

sub-zerohm, 2025-05-13:
> Any success with this?
> 
> I got 48 x 48 6S panel which has the same 4 RGB pins.

Nope.  I believe they are newer panels that use a newer hardware type that hasn't been figured out.  Going that small I think would be pointless anyway because it's too small to enjoy really.   I've built a couple much nicer displays with bigger pitch panels, I'm happy with what I can drive

closed 2025-05-13 by sub-zerohm

marcmerlin, 2025-05-13:
I'm leaving this closed, but indeed 4 color bitplanes instead of 2 would take a non trivial patch to the code to support it, and also more I/O pins since this is not HUB75 anymore, so you'd have to make your own adapter board.
As you said, not worth it unless you need bright outdoor panels in which case /6 is obviously better than /16 or /32 for brightness.