hzeller/rpi-rgb-led-matrix #1670: terrible doubt

fragmation, 2024-06-29:
as I put on the tittle I have the doutb if it can run 9 matrix (192 pixel h/w) its possible ? to make a square canvas for gift

hzeller, 2024-06-29:
It should work, bigger panesls might result in lower refresh rate.

Even before you have the hardware together, you can play with the expected refresh rate the library would output if you give all the options like `--led-chain`, `--led-parallel` but also the relevant panel type , row address type and multiplexing that are specific to your panels.

Then also give `--led-show-refresh`.
Also play with various `--led-slowdown-gpio` values (some panels can't get the data fast enough).

fragmation, 2024-06-29:
thank you very much for that quick answer :D the panels are p3 64x64 rgb 1/32 scan (I think)

marcmerlin, 2024-07-05:
this should answer your question, yes you can, I did much bigger: https://marc.merlins.org/perso/arduino/post_2020-03-13_RGB-Panels_-from-192x80_-to-384x192_-to-384x256-and-maybe-not-much-beyond.html

closed 2025-11-27 by marcmerlin