hzeller/rpi-rgb-led-matrix #1671: Python bindings no longer build with 3.12

4censord, 2024-06-30:
When trying to build the python buildings with python3.12, one gets the following errors

<details>
<summary>The build error</summary

make -C ../../lib
make[1]: Entering directory 'rpi-rgb-led-matrix/lib'
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o gpio.o gpio.cc
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o led-matrix.o led-matrix.cc
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o options-initialize.o options-initialize.cc
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o framebuffer.o framebuffer.cc
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o thread.o thread.cc
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o bdf-font.o bdf-font.cc
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o graphics.o graphics.cc
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o led-matrix-c.o led-matrix-c.cc
cc  -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -c -o hardware-mapping.o hardware-mapping.c
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o pixel-mapper.o pixel-mapper.cc
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o multiplex-mappers.o multiplex-mappers.cc
g++ -I../include -W -Wall -Wextra -Wno-unused-parameter -O3 -g -fPIC  -DDEFAULT_HARDWARE='"regular"' -fno-exceptions -std=c++11 -c -o content-streamer.o content-streamer.cc
ar rcs librgbmatrix.a gpio.o led-matrix.o options-initialize.o framebuffer.o thread.o bdf-font.o graphics.o led-matrix-c.o hardware-mapping.o pixel-mapper.o multiplex-mappers.o content-streamer.o
g++ -shared -Wl,-soname,librgbmatrix.so.1 -o librgbmatrix.so.1 gpio.o led-matrix.o options-initialize.o framebuffer.o thread.o bdf-font.o graphics.o led-matrix-c.o hardware-mapping.o pixel-mapper.o multiplex-mappers.o content-streamer.o -lpthread  -lrt -lm -lpthread
make[1]: Leaving directory '/home/daniel/Documents/Projekte/rpi-rgb-led-matrix/lib'
python setup.py build --build-lib .
rgbmatrix/core.cpp: In function ‘PyObject* __pyx_pf_9rgbmatrix_4core_6Canvas_4__reduce_cython__(__pyx_obj_9rgbmatrix_4core_Canvas*)’:
rgbmatrix/core.cpp:1308:65: warning: ‘PyDictObject::ma_version_tag’ is deprecated [-Wdeprecated-declarations]
 1308 | #define __PYX_GET_DICT_VERSION(dict)  (((PyDictObject*)(dict))->ma_version_tag)
      |                                                                 ^~~~~~~~~~~~~~
rgbmatrix/core.cpp:905:43: note: in definition of macro ‘likely’
  905 |   #define likely(x)   __builtin_expect(!!(x), 1)
      |                                           ^
rgbmatrix/core.cpp:1336:43: note: in expansion of macro ‘__PYX_GET_DICT_VERSION’
 1336 |     (var) = (likely(__pyx_dict_version == __PYX_GET_DICT_VERSION(__pyx_d))) ?\
      |                                           ^~~~~~~~~~~~~~~~~~~~~~
rgbmatrix/core.cpp:3148:5: note: in expansion of macro ‘__Pyx_GetModuleGlobalName’
 3148 |     __Pyx_GetModuleGlobalName(__pyx_t_4, __pyx_n_s_pyx_unpickle_Canvas); if (unlikely(!__pyx_t_4)) __PYX_ERR(1, 13, __pyx_L1_error)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/include/python3.12/dictobject.h:90,
                 from /usr/include/python3.12/Python.h:64,
                 from rgbmatrix/core.cpp:6:
/usr/include/python3.12/cpython/dictobject.h:22:34: note: declared here
   22 |     Py_DEPRECATED(3.12) uint64_t ma_version_tag;
      |                                  ^~~~~~~~~~~~~~
rgbmatrix/core.cpp:1308:65: warning: ‘PyDictObject::ma_version_tag’ is deprecated [-Wdeprecated-declarations]
 1308 | #define __PYX_GET_DICT_VERSION(dict)  (((PyDictObject*)(dict))->ma_version_tag)
      |                                                                 ^~~~~~~~~~~~~~
rgbmatrix/core.cpp:905:43: note: in definition of macro ‘likely’
  905 |   #define likely(x)   __builtin_expect(!!(x), 1)
      |                                           ^
rgbmatrix/core.cpp:1336:43: note: in expansion of macro ‘__PYX_GET_DICT_VERSION’
 1336 |     (var) = (likely(__pyx_dict_version == __PYX_GET_DICT_VERSION(__pyx_d))) ?\
      |                                           ^~~~~~~~~~~~~~~~~~~~~~
rgbmatrix/core.cpp:3148:5: note: in expansion of macro ‘__Pyx_GetModuleGlobalName’
 3148 |     __Pyx_GetModuleGlobalName(__pyx_t_4, __pyx_n_s_pyx_unpickle_Canvas); if (unlikely(!__pyx_t_4)) __PYX_ERR(1, 13, __pyx_L1_error)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/python3.12/cpython/dictobject.h:22:34: note: declared here
   22 |     Py_DEPRECATED(3.12) uint64_t ma_version_tag;
      |                                  ^~~~~~~~~~~~~~
rgbmatrix/core.cpp:1308:65: warning: ‘PyDictObject::ma_version_tag’ is deprecated [-Wdeprecated-declarations]
 1308 | #define __PYX_GET_DICT_VERSION(dict)  (((PyDictObject*)(dict))->ma_version_tag)
      |                                                                 ^~~~~~~~~~~~~~
rgbmatrix/core.cpp:905:43: note: in definition of macro ‘likely’
  905 |   #define likely(x)   __builtin_expect(!!(x), 1)
      |                                           ^
rgbmatrix/core.cpp:1336:43: note: in expansion of macro ‘__PYX_GET_DICT_VERSION’
 1336 |     (var) = (likely(__pyx_dict_version == __PYX_GET_DICT_VERSION(__pyx_d))) ?\
      |                                           ^~~~~~~~~~~~~~~~~~~~~~
rgbmatrix/core.cpp:3148:5: note: in expansion of macro ‘__Pyx_GetModuleGlobalName’
 3148 |     __Pyx_GetModuleGlobalName(__pyx_t_4, __pyx_n_s_pyx_unpickle_Canvas); if (unlikely(!__pyx_t_4)) __PYX_ERR(1, 13, __pyx_L1_error)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/python3.12/cpython/dictobject.h:22:34: note: declared here
[... 296 lines cut ...]
      |                                                       ^~~~~~~~
rgbmatrix/core.cpp: In function ‘long int __Pyx_PyInt_As_long(PyObject*)’:
rgbmatrix/core.cpp:12306:55: error: ‘PyLongObject’ {aka ‘struct _longobject’} has no member named ‘ob_digit’
12306 |             const digit* digits = ((PyLongObject*)x)->ob_digit;
      |                                                       ^~~~~~~~
rgbmatrix/core.cpp:12361:55: error: ‘PyLongObject’ {aka ‘struct _longobject’} has no member named ‘ob_digit’
12361 |             const digit* digits = ((PyLongObject*)x)->ob_digit;
      |                                                       ^~~~~~~~
rgbmatrix/core.cpp: In function ‘size_t __Pyx_PyInt_As_size_t(PyObject*)’:
rgbmatrix/core.cpp:12578:55: error: ‘PyLongObject’ {aka ‘struct _longobject’} has no member named ‘ob_digit’
12578 |             const digit* digits = ((PyLongObject*)x)->ob_digit;
      |                                                       ^~~~~~~~
rgbmatrix/core.cpp:12633:55: error: ‘PyLongObject’ {aka ‘struct _longobject’} has no member named ‘ob_digit’
12633 |             const digit* digits = ((PyLongObject*)x)->ob_digit;
      |                                                       ^~~~~~~~
rgbmatrix/core.cpp: In function ‘Py_ssize_t __Pyx_PyIndex_AsSsize_t(PyObject*)’:
rgbmatrix/core.cpp:13122:47: error: ‘PyLongObject’ {aka ‘struct _longobject’} has no member named ‘ob_digit’
13122 |     const digit* digits = ((PyLongObject*)b)->ob_digit;
      |                                               ^~~~~~~~
error: command '/usr/bin/gcc' failed with exit code 1
make: *** [Makefile:67: build-python] Error 1

</details>

I sadly have 0 idea how python bindings work, so i probably wont be able to create a patch myself.

yaturner, 2024-07-04:
I have the same problem, other than build and install python 2.7.11 is there a solution? Please help

4censord, 2024-07-04:
you can install any older version of python, e.g. 3.11 works fine.

yaturner, 2024-07-04:
Thanks for the tip! but I am using python3.11.2 and the bindings/python
will not build

were you able to build the bindings?

RGreinacher, 2024-07-08:
I can report the same issue with 3.12.1

neuschaefer, 2024-07-09:
The bindings need to be re-generated with cython. On Debian/Ubuntu, this would be:

```sh
sudo apt-get install cython3-legacy   # install cython 0.29.x, not 3.0.x
make -C bindings/python/rgbmatrix -B CYTHON=cython3
make
```

closed 2024-07-10 by hzeller