hzeller/rpi-rgb-led-matrix #1681: [Q&A] Dot Correction available at some point?

bluelasers, 2024-07-10:
I do not know if you plan to add this to the application interface or to the library at some point. It seems to me this is important for LED matrixes.

bluelasers, 2024-07-10:
What is the state of the wiki page here? Is this worth any time, if I created a few sections to brain dump in?

hzeller, 2024-07-11:
Dot correction is currently not planned and probably the set of users using this library (fun little blinky LED hacks using commodity Raspberry Pi) is sufficiently different from the set of users who need dot correction (color-accurate movie output) that it warrants the extra complexity.

I think the wiki page is not super-actively maintained.

closed 2024-07-11 by hzeller