hzeller/rpi-rgb-led-matrix #1682: Error loading font using python venv

sperrow, 2024-07-12:
The current "Bookworm" version of Raspberry Pi OS now requires Python Virtual Environments, so I've been trying to update my app which uses this library. I've set up and activated a venv in my project folder but I'm getting this error:

```
spotipi[28551]:   File "/home/sperrow/spotipi-lyrics/python/matrixText.py", line 61, in setLanguage
spotipi[28551]:   File "graphics.pyx", line 32, in rgbmatrix.graphics.Font.LoadFont
spotipi[28551]: Exception: Couldn't load font /home/sperrow/spotipi-lyrics/python/fonts/4x6.bdf
```

Has anyone encountered this issue? My app works fine on an older OS but I would like to update it for the new requirements going forward.

davemaster, 2024-07-20:
This is a python issue. Do this, do a list of all libraries installed where your app is working, then:
```

pip3 freeze > requirements.txt  # Python3
pip freeze > requirements.txt  # Python2 
```

Once you have the requirements.txt; activate your environment, and execute:
```
pip install -r requirements.txt
pip3 install -r requirements.txt
```

Then try....

Best regards

hzeller, 2024-07-20:
note, that by default, the 'root' permission will be dropped to 'daemon' after the hardware is initialized.
So make sure your font has readability for everyone `chmod a+r 4x6.bdf`

sperrow, 2024-07-24:
@hzeller thank you for the response and sorry for my incomplete explanation of the problem. To clarify, my app is a systemd service that runs a python script using this library. I've made a sample repo to demonstrate the issue: https://github.com/sperrow/test-rpi-rgb-led-matrix

I used [this adafruit guide](https://learn.adafruit.com/adafruit-rgb-matrix-plus-real-time-clock-hat-for-raspberry-pi/driving-matrices) to install the library (running a RPi 3 with the adafruit bonnet), then my setup script creates the python venv and systemd service.

The script `runtext.py` is copied from the bindings/python/samples folder.

Error:
```
Jul 23 19:46:17 three testmatrix[4862]: Traceback (most recent call last):
Jul 23 19:46:17 three testmatrix[4862]:   File "/home/matthewsperry/Workspace/test-rpi-rgb-led-matrix/runtext.py", line 39, in <module>
Jul 23 19:46:17 three testmatrix[4862]:   File "/home/matthewsperry/Workspace/test-rpi-rgb-led-matrix/samplebase.py", line 76, in process
Jul 23 19:46:17 three testmatrix[4862]:   File "/home/matthewsperry/Workspace/test-rpi-rgb-led-matrix/runtext.py", line 20, in run
Jul 23 19:46:17 three testmatrix[4862]:   File "graphics.pyx", line 32, in rgbmatrix.graphics.Font.LoadFont
Jul 23 19:46:17 three testmatrix[4862]: Exception: Couldn't load font /home/matthewsperry/Workspace/test-rpi-rgb-led-matrix/fonts/4x6.bdf
```

Permission:
```
matthewsperry@three:~/Workspace/test-rpi-rgb-led-matrix/fonts $ ls -l 7x13.bdf
-rw-r--r-- 1 matthewsperry matthewsperry 400760 Jul 23 18:01 7x13.bdf
```

Since my app works correctly without the venv I would guess the issue is there, but I don't have a lot of experience with python or systemd so apologies if I'm missing something more obvious.

hzeller, 2024-07-24:
Maybe any of the directories can not be changed into by user daemon ?

```
sudo -u daemon ls -l /home/matthewsperry/Workspace/test-rpi-rgb-led-matrix/fonts/4x6.bdf
```

Make sure to `chmod a+rx` on every directory /home, /home/matthewsperry,  /home/matthewsperry/Workspace/ ... to make sure user `daemon` can change into it.

Alternatively, you can load the font _before_ you start the matrix.
Or you change the options so that you don't drop permissions.

sperrow, 2024-07-24:
Not too familiar with the daemon user but my understanding is a systemd service runs as root by default? https://www.freedesktop.org/software/systemd/man/latest/systemd.exec.html#User=

I do see permission denied when using `sudo -u daemon ls`, so I tried chmod on every directory but that doesn't seem to make a difference:
```
matthewsperry@three:~ $ chmod a+rx /home/matthewsperry/Workspace/
matthewsperry@three:~ $ sudo -u daemon ls -l /home/matthewsperry/
...
ls: cannot access '/home/matthewsperry/Workspace': Permission denied
```

hzeller, 2024-07-24:
looks like you need to apply the chmod a+rx as described above

```
sudo chmod a+rx /home
sudo chmod a+rx /home/matthewsperry
sudo chmod a+rx /home/matthewsperry/Workspace
```

sperrow, 2024-07-25:
Ah thanks I did run it on every folder but didn't use sudo, now it can view it:
```
matthewsperry@three:~ $ sudo -u daemon ls -l /home/matthewsperry/Workspace/test-rpi-rgb-led-matrix/fonts/7x13.bdf
-rwxr-xr-x 1 matthewsperry matthewsperry 400760 Jul 23 18:01 /home/matthewsperry/Workspace/test-rpi-rgb-led-matrix/fonts/7x13.bdf
```
The error is still the same though
```
Jul 24 00:08:34 three testmatrix[2286]: Traceback (most recent call last):
Jul 24 00:08:34 three testmatrix[2286]:   File "/home/matthewsperry/Workspace/test-rpi-rgb-led-matrix/runtext.py", line 39, in <module>
Jul 24 00:08:34 three testmatrix[2286]:     if (not run_text.process()):
Jul 24 00:08:34 three testmatrix[2286]:             ^^^^^^^^^^^^^^^^^^
Jul 24 00:08:34 three testmatrix[2286]:   File "/home/matthewsperry/Workspace/test-rpi-rgb-led-matrix/samplebase.py", line 76, in process
Jul 24 00:08:34 three testmatrix[2286]:     self.run()
Jul 24 00:08:34 three testmatrix[2286]:   File "/home/matthewsperry/Workspace/test-rpi-rgb-led-matrix/runtext.py", line 20, in run
Jul 24 00:08:34 three testmatrix[2286]:     font.LoadFont(font_path)
Jul 24 00:08:34 three testmatrix[2286]:   File "graphics.pyx", line 32, in rgbmatrix.graphics.Font.LoadFont
Jul 24 00:08:34 three testmatrix[2286]: Exception: Couldn't load font fonts/7x13.bdf
```
A couple of things I've since noticed:
1. I assume I was wrong about the venv making a difference (works with or without it from the command line, doesn't work with or without it from systemd) so I removed it from my test repo.
2.  When running `sudo python3 runtext.py` from the command line, it loads the font but only if I use a relative path "fonts/7x13.bdf". If I use an absolute path it gives the same error as the systemd version:
```
Traceback (most recent call last):
  File "/home/matthewsperry/Workspace/test-rpi-rgb-led-matrix/runtext.py", line 39, in <module>
    if (not run_text.process()):
            ^^^^^^^^^^^^^^^^^^
  File "/home/matthewsperry/Workspace/test-rpi-rgb-led-matrix/samplebase.py", line 76, in process
    self.run()
  File "/home/matthewsperry/Workspace/test-rpi-rgb-led-matrix/runtext.py", line 20, in run
    font.LoadFont(font_path)
  File "graphics.pyx", line 32, in rgbmatrix.graphics.Font.LoadFont
Exception: Couldn't load font /home/matthewsperry/Workspace/test-rpi-rgb-led-matrix/fonts/4x6.bdf
```

I read through #1170 which was helpful to understand the root/daemon switch, also tried using drop_privileges=False but it's still the same result.

hzeller, 2024-07-25:
Did you run with strace to see what is going on?

hzeller, 2024-07-25:
(I am wondering why you make your life so hard using Python when there is a c++ example that is easier to understand and possibly is more robust ?)

sperrow, 2024-07-25:
Finally got it to work! It was the permission issue like you said, exacerbated by a separate typo in the font name that made it take longer to verify. 🥲

I've seen comments that the default permission settings changed in bookworm but I haven't found a doc to confirm that, either way I think I have a solution for now. Thank you for all your help and for building this library!

(As for why python, I built a [small project](https://sperrow.cc/2024/05/09/spotipi-lyrics.html) using the Spotify python lib. I'm primarily a front end dev so all of this was new to me and I assume learning python is easier than c++ 😅)

closed 2024-07-25 by sperrow

hzeller, 2024-07-25:
Yay! Have fun with your (now working) project!