hzeller/rpi-rgb-led-matrix #1683: Problem turning off audio in /boot.congig.txt - Pi 4 Model B

ElBertrando, 2024-07-12:
I have a Pi 4 Model B running Debian 12. Upon installing the software for the Adafruit Matirx Hat + RTC and testing some the examples in the subdirectory examples-api-use, I find that I need to turn off the sound module:
[image]
However when I go to edit the parameter in `/boot/config.txt`, it says the file has moved to` /boot/firmware.txt`
[image]
When I edit that file, it has no effect on the example code -- it seems does not recognize it:
[image]
I have rebooted the pi several times and yet the I still get the error mesage when running the example code...

davemaster, 2024-07-13:
use: 

sudo raspi-config, and do from there

ElBertrando, 2024-07-15:
I tried rapsi-config to avail but was able to turn it off with a combination of turning it off in ` /boot/firmware.txt` and adding the line `blacklist snd_bcm2835` by : `cd /etc/modprobe.d`    then ---> `sudo nano alsa-blacklist.conf` 

I can close issue now.. thx

closed 2024-07-15 by ElBertrando