hzeller/rpi-rgb-led-matrix #1685: 64 x 64 LED matrix

TristanGrahn, 2024-07-19:
Hey all, sorry I know this is not the right place to post this, but I have been struggling for almost two days straight.

long story short I bought a p3(2121) 64x64 1/32S waveshare 
(https://www.waveshare.com/rgb-matrix-p3-64x64.htm)

it came with a hub75 to jumper pin cable where I plugged im the individual pins into a Rapsberry Pi 3B according the the pinout provided. (I’ve re-wired and checked the wireing multiple times)

the raspberry pi is running rasbian lite x64 where I cloned this library and blacklisted the audio module.

my problem is no matter what combination of flag variables I choose for the “demo -D0” program (panel type, multiplexing, row-added-type) the panel just stays blank not even a flicker.

(The panel does work when I connect an esp32 to it) 

Is there a specific thing I need to do or configure in the boot file or a specific flag varible I’m not using?
 I just feel like I’ve tried everything.
[image]

davemaster, 2024-07-19:
Greetings,

And, what is your command? Your power supply? How much Amps gives? ONE MORE, Maybe you're swap the wiring HUB75 pinout, maybe...

TristanGrahn, 2024-07-20:
Sorry for the vague description. I do apologise 
The display is being powered with a 5V 6A power supply and the raspberry pi is being powered by a 5V 3A power supply (the grounds of both supplies are connected together. 

I did double check that I didn’t swap the hub75 wireing and all seems correct.

the command I was using was:

Sudo ./ -D0 —led-rows=64 —led-cols=64 —led-slowdown-gpio=4 —led-chain=1  --led-gpio-mapping=regular

I’ve also tried the two different display types 
Fm6126A and FM6217 but the display still stays blank.
I also disabled 1-wire and I2S

Could the length of the cable from the pi to the display be a problem?

LyubMaks, 2024-07-20:
Hello.  what chips are used in the panel?  I worked with a similar one, only at the end of the marking -M3.  it started without any problems.  try to enter the command correctly: go to the rpi-rgb-led-matrix directory.  type make.  go to the directory rpi-rgb-led-matrix/examples-api-use.  enter the command
 sudo ./demo -D0 --led-rows=64 --led-cols=64 --led-slowdown-gpio=4 --led-chain=1 --led-gpio-mapping=regular
[image]

TristanGrahn, 2024-07-20:
Hey thank you for your help I did that to no avail my board is using an SM16208SJ my end marking is M5.1 not sure what that means as well🙈 I’ve been trying things all day I fear it may have to do with the logic levels and I may need to get adafruit bonnent

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin