hzeller/rpi-rgb-led-matrix #1686: Panel P3 64x64 IC SM16208 SM5166  with matrix bonnet

LyubMaks, 2024-07-20:
Hello. I have a problem with 64x64 matrices with chips SM 6208 SM5166 scan 1/16 with strange addressing.
I made a video with an example of a pixel-mover (start x,y 0.0, then y - 0-63 and back, then x - 0-63).
Please help to make multiplexing for these panels

https://github.com/user-attachments/assets/e68b9cc2-b6b5-4e3f-b962-4b4186a380a0

https://github.com/user-attachments/assets/ac48bf31-194e-4d8d-a59d-0684579cd2f3

If you need to do additional tests, write what is needed.

LyubMaks, 2024-07-20:
[image]
[image]

LyubMaks, 2024-07-20:
I was able to run this panel in another library (DMD_STM32) by changing some parameters (screen). But this library does not fit my needs. And I don't know how to use DMD_STM32 library settings for rpi-rgb-led-matrix library.
Maybe someone can integrate it.
[image]

https://github.com/user-attachments/assets/31550e3b-d837-400c-9c8d-2abfe70563e7

TristanGrahn, 2024-07-20:
Ive also got a sm16208 p3 64x64 panel. And have been struggling to get it to display anything.

ixcue, 2024-07-22:
I have the same problem, please help. I already wrote to the author (@hzeller), but have not received a response yet...

davemaster, 2024-10-31:
Greetings, 

Have You MOD the active 3 board, to connect E-pin, or try --led-multiplexing = 0 .. to.. 18 (panel types)

davemaster, 2024-10-31:
Greetings,

Can You all share a close picture of your HUB75 connector? -INPUT-

GeorgeFlorian, 2024-11-01:
> Greetings,
> 
> Can You all share a close picture of your HUB75 connector? -INPUT-

I have made a separate issue: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1725

board707, 2024-11-22:
> I don't know how to use DMD_STM32 library settings for rpi-rgb-led-matrix library.
> Maybe someone can integrate it.

Where the led-multiplexing setting are located in this library?

marcmerlin, 2024-11-26:
grep the code for led-row-addr-type and see where that leads you

board707, 2024-11-26:
Thanks, I already found the file `multiplex-mappers.cc`

closed 2025-11-27 by marcmerlin