hzeller/rpi-rgb-led-matrix #1687: Error with fornt when running venv code as root user -- Similar to #1682

ElBertrando, 2024-07-22:
As venv is required for Debian 12 in my Pi 4 model B, I found I can run the code by directing the file path to the virtual env ---> `/home/pi/Code/SignEnv/bin/python3 /home/pi/Code/Main.py`. Code runs fine but I see the message in terminal to run code as root: -->**_Can't set realtime thread priority=99: Operation not permitted.
	You are probably not running as root ?
	This will seriously mess with color stability and flicker
	of the matrix. Please run as `root` (e.g. by invoking this
	program with `sudo`), or setting the capability on this
	binary by calling
	sudo setcap 'cap_sys_nice=eip' /usr/bin/python3.11_**

  However when I run as root --> `sudo /home/pi/Code/SignEnv/bin/python3 /home/pi/Code/Main.py`   there is the font error -->  
 
**_File_"graphics.pyx", line 32, in rgbmatrix.graphics.Font.LoadFont
Exception: Couldn't load font /home/pi/rpi-rgb-led-matrix/fonts/10x_20.bdf_**

I tried enabling in the font directory -->`chmod a+r 10x20.bdf ` to no avail.

hzeller, 2024-07-22:
Do you really have an underscore at the end of the filename ?

ElBertrando, 2024-07-22:
No that's a copy and past error --- I'll edit that out now

hzeller, 2024-07-22:
The Exception still shows an underscore but at a different place :)

To double-check, I'd copy the message you get from the Exception and check if the file actualy is readable (which I suspect it is not with that underscore):

`ls -l /home/pi/rpi-rgb-led-matrix/fonts/10x_20.bdf`

Also, the good old running `sudo strace -f <program>` to see what is actually going on in the system might help narrow down what is going on.

ElBertrando, 2024-07-29:
As I was on vacation, I am getting back to this issue; The underscore was a typo [I had it in two places - I've fixed the other reference].  I have verified that the file exists with  `ls -l /home/pi/rpi-rgb-led-matrix/fonts/10x20.bdf`. A stack trace by program name `sudo strace -f python3` I see the following section repeated - not sure what is means:
```
pselect6(1, [0], NULL, NULL, NULL, NULL) = ? ERESTARTNOHAND (To be restarted if no handler)
--- SIGWINCH {si_signo=SIGWINCH, si_code=SI_KERNEL} ---
rt_sigreturn({mask=[]})                 = -1 EINTR (Interrupted system call)
ioctl(0, TIOCGWINSZ, {ws_row=28, ws_col=111, ws_xpixel=85, ws_ypixel=0}) = 0
>>> )            = 9
```

hzeller, 2024-07-29:
The question is if not your user but the drop privilege user can read the file, so you need to switch to that user first:

```bash
sudo -u daemon ls -l /home/pi/rpi-rgb-led-matrix/fonts/10x20.bdf

# Or even better to see if the content can be read
sudo -u daemon cat /home/pi/rpi-rgb-led-matrix/fonts/10x20.bdf
```

ElBertrando, 2024-07-29:
Interesting -- even as root Permission denied:

[image]

hzeller, 2024-07-29:
not as root, as `daemon`, as you switch to be that user (look at that command line again)

This is why I am saying you need to make sure that all directories in the path can be changed into by the user

Something like

```bash
for f in /home /home/pi /home/pi/rpi-rgb-led-matrix /home/pi/rpi-rgb-led-matrix/fonts ; do
   chmod a+rx $f
done
chmod a+r /home/pi/rpi-rgb-led-matrix/fonts/10x20.bdf
```

ElBertrando, 2024-07-29:
I tried this but : 

[image]

However this did work:

`sudo chmod a+r /home/pi/rpi-rgb-led-matrix/fonts`

hzeller, 2024-07-29:
so, you got it to work ?

ElBertrando, 2024-07-29:
Yes -- thx for all of the help -- I will close it now

closed 2024-07-29 by ElBertrando