hzeller/rpi-rgb-led-matrix #1692: Need help with audio modules

vsumpi, 2024-07-30:
Hi, my project needs audio not High-definition but a loudspeaker or buzzer with a low/high frequency. Do you have any board or USB recommendations? I will need an amp too for 4-amp 2-watt speakers. the panel occupies almost the whole GPIO output so my only choice for a Rpi4 is USB.

davemaster, 2024-08-13:
in your command, you can use --led-hardware-no-pulse, then you have no need to disable the sound in your Raspberry Pi 3 from the boot. I use the sound, a speaker with amp plugged in the audio of my [Rpi 3, and works great](https://youtu.be/DsmSmaUgFRw)

vsumpi, 2024-08-16:
> amp plugged in the audio
Is it a 3.5mm jack or USB?

davemaster, 2024-08-16:
It is the 3.5mm jack in a RPi 3 B+

davemaster, 2024-08-21:
This was the usb sound card

Modelo: USB Sound Card Adapter 7.1 Channel
[image]

closed 2024-08-23 by vsumpi