hzeller/rpi-rgb-led-matrix #1693: c#

Dasca71, 2024-08-01:
Hi,
I have tried several times to install .NET 6 to run the sample projects but without success. Can someone guide me on the correct way to install .NET on a Raspberry Pi running the DietPi distribution?

thx in advance

Davide

davemaster, 2024-08-01:
Greetings,

You mean, C# for raspberry pi? What environment? link? I'm very interested and together we can take this out

davemaster, 2024-08-01:
> Hi, I have tried several times to install .NET 6 to run the sample projects but without success. Can someone guide me on the correct way to install .NET on a Raspberry Pi running the DietPi distribution?
> 
> and of course this command sudo apt install dotnet6 don't work
> 
> thx in advance
> 
> Davide

What about this link?
[THIS LINK ->](
https://learn.microsoft.com/es-es/dotnet/iot/deployment)

ivancmz, 2025-09-16:
Yes, as @davemaster said (a year ago) that's the link: https://learn.microsoft.com/en-us/dotnet/iot/deployment

Also, my two cents, you can use --channel LTS for the Long Term Support version (currently .net 8) or a specific version with --version:

```
curl -sSL https://dot.net/v1/dotnet-install.sh | bash /dev/stdin --version 8.0.413
echo 'export DOTNET_ROOT=$HOME/.dotnet' >> ~/.bashrc
echo 'export PATH=$PATH:$HOME/.dotnet' >> ~/.bashrc
source ~/.bashrc
```
and then check with:

`dotnet --version`