hzeller/rpi-rgb-led-matrix #1696: 64x64 panel, every pair of lines are blank

Tenumen, 2024-08-06:
Hi there
Would someone be able to point me to what is wrong with my setup and causing this issue where two lines are working then the next two are blank?
[image]
I am using an unnamed panel with E but no D channel. I obtained this pattern by running the Plasma Pattern example with only the pin-outs changed. I am using an ESP32-WROOM Dev board connected directly to the panel via a simple breadboard -so I am sending 3.3v signals to the panel.
[image]
Any guidance that someone can offer me would be much appreciated.
Thanks

hzeller, 2024-08-06:
This project is using a Raspberry Pi, not an ESP32.

Tenumen, 2024-08-07:
Oops sorry! I've been looking at both yours and https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA and got the two mixed up.

closed 2025-11-27 by marcmerlin