hzeller/rpi-rgb-led-matrix #1697: ArchLinux arm working, but message at startup

mrechte, 2024-08-07:
Hello.

This what I get when running the demo on a rpi3 b+ with latest Archlinux :
```
# ./demo --led-cols=64 -D 1 runtext.ppm
non-existent Revision: Could not determine Pi model
Failed to read revision from /proc/device-tree/system/linux,revision
Unknown Revision: Could not determine Pi model
Suggestion: to slightly improve display update, add
	isolcpus=3
at the end of /boot/cmdline.txt and reboot (see README.md)
Size: 64x32. Hardware gpio mapping: regular
Read image 'runtext.ppm' with 500x32
Press <CTRL-C> to exit and reset LEDs
```
Note that the demo is running okay.
There is no system subdir:
```
# ls /proc/device-tree/
'#address-cells'   chosen       cpus		       leds	    model   reserved-memory   soc	      timer
 aliases	   clocks       framebuffer@3e99a000   memory@0     name    serial-number     __symbols__     wifi-pwrseq
 arm-pmu	   compatible   interrupt-parent       memreserve   phy    '#size-cells'      thermal-zones

# uname -a
Linux pi3 6.9.12-2-aarch64-ARCH #1 SMP PREEMPT_DYNAMIC Sat Jul 27 13:48:15 MDT 2024 aarch64 GNU/Linux
```

hzeller, 2024-08-07:
Can you figure out what the equivalent way of finding out on Arch is ?

mrechte, 2024-08-08:
Sure: 
```
cat /proc/device-tree/model 
Raspberry Pi 3 Model B Plus Rev 1.4
```

hzeller, 2024-08-08:
That is the human readable form, but there should be a way to read the binary value describing the Pi revision from 
`/proc/cpuinfo` or `/proc/device-tree/system/linux,revision`.
Probably would be good if Arch adopted the same device-tree access as the official Pi kernels or provide another way to  read the pi revision ( https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#raspberry-pi-revision-codes )

mrechte, 2024-08-08:
OK I found this thread on [ArchLinux ARM forum](https://archlinuxarm.org/forum/viewtopic.php?f=67&t=13425&p=60701&hilit=revision#p60701)

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828