hzeller/rpi-rgb-led-matrix #1700: Flickering at regular intervals

Dowanproject, 2024-08-23:
@hzeller Thanks for your help. I've been testing it for a few weeks. Noticeable flicker is almost eliminated by setting gpio-slowdown, pwm-lsb-nanoseconds, pwm-bits, but subtle flicker still occurs, especially if the colors are complex.

./video-viewer --led-rows=64 --led-cols=128 --led-chain=2 --led-parallel=3 --led-slowdown-gpio=5 --led-pwm-bits=8 --led-pwm-lsb-nanoseconds=90 -T2 --led-no-drop-privs /dev/video0

hzeller, 2024-08-23:
Wouldn't it be better to continue on #1689 instead of creating a new issue ?

You're setting the nanoseconds pretty low and led-pwm-bits pretty low which can be problematic as you're maxing out CPU.
I'd probably keep the led-pwm-bits as-is and rather play with `--led-pwm-dither-bits` to get higher refresh.

Also, use `--led-show-refresh` to determine your current refresh rate and its fluctuations (it will print the lowest observed as well), and then set a `--led-limit-refresh` to something reasonable. Say you see a refresh of 180Hz with glitces down to 146Hz', maybe set it to 150Hz.

In general, make sure to configure your `/dev/vido0` output to whatever is closest to the target resolution to not waste too much CPU to scale.

In general, you're pushing a lot of pixels, so I wouldn't be surprised if you would get some slight flicker.

'regular intervals' sounds like something is running regularly on your machine (maybe having a 'top' kept running ? Not runnging a minimal no-GUI distribution ? etc.).

marcmerlin, 2024-11-15:
closing the dupe bug

closed 2024-11-15 by marcmerlin