hzeller/rpi-rgb-led-matrix #1704: 128x64 P2 led matrix not working

raulzanardo, 2024-08-30:
Hello guys!

I just got two of this 128x64 P2 (P2-1515-128x64-32s-S2) led metrixes from AliExpress and after testing with all the settings I could not make ir work, both of them. Not even random colors and lines, just black. I teste others led matrixes what I have with my setup and everything worked fine, just the new one that doesn't work. 

https://pt.aliexpress.com/item/1005007309542250.html?spm=a2g0o.order_list.order_list_main.11.3dad194dImMZro&gatewayAdapt=glo2bra

The chips are:
6363C NET23800C
6363C NET23900C
DP32019B 312501
And some 74hc254 and 74hc04 as usual. 

Do you guys now this matrix and how to make it work?

[image]
[image]
[image]
[image]

davemaster, 2024-08-30:
Greetings, 

You're trying this first time? Ok, according vendor, this is a 32S panel... You know what that means?

raulzanardo, 2024-08-31:
Hi! 
I've already used 64x64 1/32. I have an raspberry pi 3b+ with 4 64x64 1/32 in 2 series 2 pararellel mode. And two others 64x64 1/32 with ESP32. 

There is as difference between 1/32 and 32s led matrixes? As I understand, the 1/32 means that you have to use the A, B, C, D and E pins, and normaly the E pin is not exposed in some boards, but i'm using a direct (regular) adapter.

My pinout is the regular one, as described on the link you shared.

The command line I used was this (sorry, I forgot o share before):

`sudo ~/apps/rpi-rgb-led-matrix/examples-api-use/demo \
-D0 \
--led-pwm-dither-bits=0 \
--led-gpio-mapping=regular \
--led-brightness=60 \
--led-rows=64 \
--led-cols=128 \
--led-chain=1 \
--led-parallel=1 \
--led-slowdown-gpio=2 \
--led-show-refresh \
--led-scan-mode=0 \
--led-limit-refresh=90 \
--led-pwm-bits=11`

I've play arround with --led-slowdown-gpio , --led-pwm-bits , --led-row-addr-type , --led-pwm-lsb-nanoseconds,  --led-pwm-dither-bits,  --led-no-hardware-pulse, --led-panel-type. 

The vendor just says that the IC is 6238.

davemaster, 2024-09-01:
Ok, only the power issue is left... your panel is like x4 panels... minimum of 5V 20A power supply (just for try) and second, try like the panel were a 64x64 panel, just for see some output

My experience, double check each of those cables.... You did? Do it again..

raulzanardo, 2024-09-02:
I'm using a 10A 5A power supply that works fine with the others 64x64 and also with 4x 64x64 setup I have. I'm also checking the voltage while I test it. 
This setup works fine with other 64x64 using the same cable, power supply and adapter. Already check everything but will do it again. I even test it with an ESP32 and the esp32-hub75-matrixpanel-i2s-dma  library setup I have. Its really weird.  

This week I will try to do some bit bang and use a oscilloscope to se what is happening.

davemaster, 2024-09-02:
Good alternative using the esp32-hub75-matrixpanel-i2s-dma library for discard a rgb panel malfunction. Can You share video of your panel working with the esp32 ?

raulzanardo, 2024-09-02:
With the esp32-hub75-matrixpanel-i2s-dma library the 128x64 didn't work. 
But I will share any progress I make.

davemaster, 2024-09-02:
> With the esp32-hub75-matrixpanel-i2s-dma library the 128x64 didn't work. But I will share any progress I make.

I suggest You, to try [this 128x64 Project](https://github.com/bogd/esp32-morphing-clock), So You can probe your new panel. Some times, vendors sent broken or untested parts... lets check that out

[image]

raulzanardo, 2024-09-02:
nice! Thanks, I will try this one!

davemaster, 2024-09-02:
> nice! Thanks, I will try this one!

Me too, but with my 64x64 panels...  I don't have those 128x64 yet

raulzanardo, 2024-09-02:
The time I try to use two 64x64 with the ESP32 I ran out of memory haha but it worked for simple things and withou double buffer. 

Is hard to buy these led matrix from AliExpress, I tried to make a 6x 64x64 once and the last two I bought came different. They never send the same one.

davemaster, 2024-09-02:
> Is hard to buy these led matrix from AliExpress, I tried to make a 6x 64x64 once and the last two I bought came different. They never send the same one.

The BEST way, is be patient, and wait for some review, or buyer comments. Then buy those probed working panels

raulzanardo, 2024-09-08:
The vendor said that the panel use this chips:
[image]

davemaster, 2024-09-11:
> The time I try to use two 64x64 with the ESP32 I ran out of memory haha but it worked for simple things and withou double buffer.
> 
> Is hard to buy these led matrix from AliExpress, I tried to make a 6x 64x64 once and the last two I bought came different. They never send the same one.

Then, try an example and been 32S, program like if You have a 64x32 32S RGB Moduel.... what happens?

board707, 2024-10-14:
> The vendor said that the panel use this chips: 6238 + 7262

Is that the same panel that shown on the photos in your first post? It is clearly viewable that the chips on the panel are FM6363C and DP32019.
The driver FM6363 is a S-PWM type and must be managed with custom procedure, incompatible with conventional HUB75 code.

raulzanardo, 2024-10-14:
> > The vendor said that the panel use this chips: 6238 + 7262
> 
> Is that the same panel that shown on the photos in your first post? 
Yes

>It is clearly viewable that the chips on the panel are FM6363C and DP32019. The driver FM6363 is a S-PWM type and must be managed with custom procedure, incompatible with conventional HUB75 code. See issue #466 for more info

Ok, thanks!

closed 2025-11-27 by marcmerlin