hzeller/rpi-rgb-led-matrix #1705: Trying to get Started

utemorley2012, 2024-09-03:
Hi All,

Pretty new to Matrix LED Panels and Raspberry Pi but I am attempting to build my own sports ticker. I have 5 128x64 modules I am trying to set in parallel with a raspberry pi 4. I have followed all of the tutorials in the READMEs but just can't seem to get the panels to do anything really. I have checked the wiring (regular mapping) again and again and am wondering if there is a way to just test to see if the panels are working. I only have 1 connected right now and confirmed it is receiving 5.15 V. Here is an image of my panels. 
[image]

I have tried lots of different demo code option combos (too many to list here) and still nothing. Any ideas? Thanks so much!

NathanBeesley, 2024-09-25:
I had some issues getting things working too. What are you using for a matrix and controller? Are you headless and where are you attempting to run the code?

Providing some more details might help.

davemaster, 2024-09-26:
Greetings,

1. If You wanna to get started, just READ, first, READ the README.md of this repo. 
2. Then, once you read, try to [choose your wiring](https://github.com/hzeller/rpi-rgb-led-matrix/blob/0ff6a6973f95d14e3206bcef1201237097fa8edd/wiring.md), you have 64 rows, YOU NEED wire with E pin (don't know, READ?)
3. THIS IS where you begin to try with the code. What RPi are You going to use: RP1, 2, 3, 4; RPi 5 can't be used YET; so from 3 to 4 RPi, You must use a section of README.md, the [slowdown issue](https://github.com/hzeller/rpi-rgb-led-matrix?tab=readme-ov-file#gpio-speed), remember, according to you RPi, and check what wiring or hat-board or whatever are you using to interface the [RPi GPIO and your RGB panel](https://github.com/hzeller/rpi-rgb-led-matrix?tab=readme-ov-file#panel-connection); 
4. When you get here, let me know...

Best regards

closed 2024-10-10 by utemorley2012